eclipse-milo/milo #815: ObjectNode#getMethod(QualifiedName browseName) fails when OutputArguments is missing

pvorb, 2021-04-07:
**Describe the bug**
I am trying to resolve a method of a `FileTypeNode` via `ObjectNode#getMethod(QualifiedName browseName)`. This operation fails with a `UaException` with status `Bad_NotFound`, because in my case the `Write` and `Close` methods of the file ObjectNode do not have the `OutputArguments` property. According to [5.7 Method NodeClass](https://reference.opcfoundation.org/v104/Core/docs/Part3/5.7/),

> [...] An empty array **or a Property that is not provided** indicates that there are no input arguments or output arguments for the Method.

(emphasis mine)

So the implementation of the OPC UA server I connect to seems to match the standard. Simply not having the property should be treated the same as if it had the property and the array was empty.

The exception originates in the class [`UaNode` at line 1035](https://github.com/eclipse/milo/blob/f33b0aed38d32700306fac6e3381ca5749aa4544/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/nodes/UaNode.java#L1035), which is called from [`UaMethodNode#readOutputArgumentsAsync()`](https://github.com/eclipse/milo/blob/f33b0aed38d32700306fac6e3381ca5749aa4544/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/nodes/UaMethodNode.java#L227-L229). I think one way to resolve the bug would be to just handle the results of the `read(Input|Output)ArgumentsAsync()` in [`UaObjectNode#getMethodAsync()`](https://github.com/eclipse/milo/blob/f33b0aed38d32700306fac6e3381ca5749aa4544/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/nodes/UaObjectNode.java#L289-L290) as follows

~~~ java
return methodNode.readInputArgumentsAsync().exceptionally(t -> new Arguments[0]).thenCombine(
    methodNode.readOutputArgumentsAsync().exceptionally(t -> new Arguments[0]),
    (in, out) -> new UaMethod(client, this, methodNode, in, out)
);
~~~

Optionally, you could also check that the exception actually has status `Bad_NotFound`. But anyway, by handling the exception, resolving and calling the method would succeed.

**Expected behavior**
`ObjectNode#getMethod(QualifiedName browseName)` should work with OPC UA servers that do not have the `InputArguments` or `OutputArguments` property on some methods.

**Additional context**
I'm using milo-client 0.6.0 against a proprietary OPC UA server that was implemented with .NET, but I'm not sure which library they are using exactly.

kevinherron, 2021-04-07:
Thank you for reporting this, I should be able to get it fixed quickly.

pvorb, 2021-04-07:
Awesome, thanks a lot!

PR #816 (2021-04-07): Fix Bad_NotFound calling methods with no InputArguments or OutputArguments
MethodNodes are allowed to omit the InputArguments and OutputArguments
PropertyNodes all together rather than using Nodes with empty values.

fixes #815
commit dec3906c (2021-04-07): Fix Bad_NotFound calling methods with no InputArguments or OutputArguments (#816)

MethodNodes are allowed to omit the InputArguments and OutputArguments
PropertyNodes all together rather than using Nodes with empty values.

fixes #815
closed 2021-04-07 by kevinherron