eclipse-milo/milo #819: 0.6.0 release dependency error

KoenR3, 2021-04-08:
**Describe the bug**
When switching to 0.6.0 I dependency error on jetbrains annotations is thrown.  This is not the case when using <= 0.5.4

I am using SBT 1.4.6, JDK 1.8

**Expected behavior**
Correct dependency resolution

**Logs and Packet Captures**
```
[error] sbt.librarymanagement.ResolveException: Error downloading org.jetbrains:annotations:RELEASE
[error]   Not found
[error]   Not found
[error]   not found: /Users/koenr/.ivy2/localorg.jetbrains/annotations/RELEASE/ivys/ivy.xml
[error]   not found: https://repo1.maven.org/maven2/org/jetbrains/annotations/RELEASE/annotations-RELEASE.pom
[error]         at lmcoursier.CoursierDependencyResolution.unresolvedWarningOrThrow(CoursierDependencyResolution.scala:258)
[error]         at lmcoursier.CoursierDependencyResolution.$anonfun$update$38(CoursierDependencyResolution.scala:227)
[error]         at scala.util.Either$LeftProjection.map(Either.scala:573)
[error]         at lmcoursier.CoursierDependencyResolution.update(CoursierDependencyResolution.scala:227)
[error]         at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:60)
[error]         at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:53)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:103)
[error]         at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:73)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$20(LibraryManagement.scala:116)
[error]         at scala.util.control.Exception$Catch.apply(Exception.scala:228)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:116)
[error]         at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:97)
[error]         at sbt.util.Tracked$.$anonfun$inputChangedW$1(Tracked.scala:219)
[error]         at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:130)
[error]         at sbt.Classpaths$.$anonfun$updateTask0$5(Defaults.scala:3486)
[error]         at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error]         at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error]         at sbt.std.Transform$$anon$4.work(Transform.scala:68)
[error]         at sbt.Execute.$anonfun$submit$2(Execute.scala:282)
[error]         at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error]         at sbt.Execute.work(Execute.scala:291)
[error]         at sbt.Execute.$anonfun$submit$1(Execute.scala:282)
[error]         at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error]         at sbt.CompletionService$$anon$2.call(CompletionService.scala:64)
[error]         at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error]         at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error]         at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] (client / update) sbt.librarymanagement.ResolveException: Error downloading org.jetbrains:annotations:RELEASE

```

**Additional context**
Version of Eclipse Milo you are using, OPC UA client or software server you are interacting with, etc...

KoenR3, 2021-04-08:
I looked at Maven and for sdk-server the version info is incorrect. It puts the version number at "updated" and RELEASE at "version". As for server it is given bith as a compile (twice even) and a provided dependency... https://mvnrepository.com/artifact/org.eclipse.milo/sdk-server/0.6.0

kevinherron, 2021-04-08:
Huh, it sure does. I don't know where that came from. I'll fix it for `0.6.1`.

kevinherron, 2021-04-08:
@KoenR3 sorry I pinned the blame on SBT at first. I had a quick look at some of the pom.xml files published to Maven and saw nothing wrong but I must have checked everything _but_ the one from `sdk-server`.

closed 2021-04-08 by kevinherron

KoenR3, 2021-04-09:
No worries, @kevinherron, SBT is often to blame for things ;-)

PR #820 (2021-04-08): Remove mistaken org.jetbrains:annotations dependencies in sdk-server
fixes #819
commit a051ae8a (2021-04-08): Remove mistaken org.jetbrains:annotations dependencies in sdk-server

fixes #819