eclipse-milo/milo #822: Use the subscription method to report a large number of errors Bad_TooManyMonitoredItems

beyondyinjl2, 2021-04-12:
The error is as follows:
failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_HMI_UNIT2_Schedule} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_HMI_UNIT2_BatchNumber} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_HMI_UNIT2_Blend} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_HMI_UNIT2_BlendID} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ1_PV_1} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ1_PV_2} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ1_PV_3} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ1_PV_5} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ1_PV_8} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ1_PV_9} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ2_PV_1} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ2_PV_2} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ2_PV_3} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ2_PV_5} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ2_PV_8} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})
16:05:04.184 ERROR [OpcuaService.java:369] com.talkweb.iot.opcua.collect.service.OpcuaService - failed to create item for nodeId=NodeId{ns=1, id=t|ZS_SZD_QSJ2_PV_9} (status=StatusCode{name=Bad_TooManyMonitoredItems, value=0x80DB0000, quality=bad})

```
<dependency>
			<groupId>org.eclipse.milo</groupId>
			<artifactId>sdk-client</artifactId>
			<version>0.5.2</version>
		</dependency>
```

Fake code:
```
        double publishInterval = 2000.0;
        if (subDevice.getCycle() > 0) {
            publishInterval = subDevice.getCycle();
        }
        UaSubscription subscription = null;
        UInteger subscriptionId = null;
        try {
            subscription = client.getSubscriptionManager().createSubscription(publishInterval).get();
            subscriptionId = subscription.getSubscriptionId();
        } catch (Exception e) {
            log.error("OPC UA Create error, {}", e.getMessage());
        }

        for (GroupDto groupDto : groupDtoList) {
            String group_name_en = groupDto.getGroup_name_en();
            GroupJsonDto groupJsonDto = fixGroupJsonDto(String.valueOf(groupDto.getGroup_json()));

            int queueSize = 10;
            if (subDevice.getOpcUaDeviceJsonDto().getQueueSize() > 0) {
                queueSize = subDevice.getOpcUaDeviceJsonDto().getQueueSize();
            }

            List<MonitoredItemCreateRequest> itemsToCreate = new ArrayList<>();
            List<ItemDto> itemDtoList = groupDto.getItems();
            for (ItemDto itemDto : itemDtoList) {
                String itemCode;
                if (!StringUtil.isNull(old_char) && !StringUtil.isNull(new_char)) {
                    itemCode = itemDto.getCode().replace(old_char, new_char);
                } else {
                    itemCode = itemDto.getCode();
                }

                String item = point_prefix + group_prefix + itemCode;

                NodeId nodeId = new NodeId(subDevice.getOpcUaDeviceJsonDto().getNamespaceIndex(), item);
                ReadValueId readValueId = new ReadValueId(nodeId, AttributeId.Value.uid(),null,null);
  
                MonitoringParameters parameters = new MonitoringParameters(
                        subscription.nextClientHandle(),    
                        publishInterval,     // sampling interval
                        null,       // filter, null means use default
                        UInteger.valueOf(queueSize),   // queue size
                        true        // discard oldest
                );
              
                MonitoredItemCreateRequest request = new MonitoredItemCreateRequest(readValueId, MonitoringMode.Reporting, parameters);
            
                itemsToCreate.add(request);
            }

            BiConsumer<UaMonitoredItem, Integer> biConsumer =
                    (item, id) -> item.setValueConsumer(new BiConsumer<UaMonitoredItem, DataValue>() {
                        @Override
                        public void accept(UaMonitoredItem uaMonitoredItem, DataValue value) {
                            log.info("subscription value received: item={}, value={}", item.getReadValueId().getNodeId(), value.getValue());

             
                            String code = getCodeMap(item.getReadValueId().getNodeId().getIdentifier().toString(),
                                    device_id, group_name_en);
                            if (code != null) {
                                Object obj = changeType(item.getReadValueId().getNodeId().getIdentifier().toString(), value.getValue().getValue());
                                putData(device_id, group_name_en,
                                        code, obj);
                            }
                        }
                    });
            List<UaMonitoredItem> items = null;
            try {
                items = subscription.createMonitoredItems(
                        TimestampsToReturn.Both,
                        itemsToCreate,
                        biConsumer
                ).get();
            } catch (Exception e) {
                log.error("OPC UA subscription error, {}", e.getMessage());
            }

            for (UaMonitoredItem item : items) {
                if (item.getStatusCode().isGood()) {
                    log.info("item created for nodeId={}", item.getReadValueId().getNodeId());
                } else {
                    log.error("failed to create item for nodeId={} (status={})", item.getReadValueId().getNodeId(), item.getStatusCode());
                }
            }
```

One connection with 30 subscribers, each subscriber has 150 items

beyondyinjl2, 2021-04-12:
The server uses wincc opcua

kevinherron, 2021-04-12:
This StatusCode means the server has a limit to the number of MonitoredItems it allows and that you have exceeded that number.

The limit could be per subscription, per session, or aggregate across all sessions and subscriptions from all clients; that is up to the server. 

I don't think there's anything else you can do from the client side.

closed 2021-04-14 by kevinherron

zcarsenal, 2021-12-15:
@beyondyinjl2

zcarsenal, 2021-12-15:
Hello, I wonder if you have solved this problem

I have the same problem

Can you tell me the solution

zcarsenal, 2021-12-15:
@kevinherron thank you very mach!