eclipse-milo/milo #829: How to delete OpcUaClient resources?

vinodhalaharvi, 2021-04-21:
We are using 0.37 Version of Milo codebase. We have OpcUaClient.create which creates a OpcUa client and all underlying data structures. But there is no `delete` counter part. Could you please guide me through the steps that's needed to perform the delete action? Pseudo code or actual code will also be very helpful.

kevinherron, 2021-04-21:
Just call `OpcUaClient::disconnect`.

closed 2021-04-21 by kevinherron

vinodhalaharvi, 2021-04-21:
hi Kevin, thanks for quick reply, after a bit of testing I do see connections getting closed  at the network level.  Just one follow up question -  Does disconnect also release in memory objects?

kevinherron, 2021-04-21:
Well... this is Java, so resources are cleaned up by the garbage collector when no longer referenced.

There is an additional cleanup you can do if you’re intending to shut down your application and not use Milo any more JVM-wide by calling `Stack.releaseSharedResources()` but this is normally only necessary in “modular” applications that may load and unload ClassLoaders as part of a module or plugin system.

vinodhalaharvi, 2021-04-21:
Great thanks, I see, looks like the disconect method dereferences the objects.

The second use case does not apply as we still need to keep milo around in the JVM for the newer client connections.