eclipse-milo/milo #836: add support for missing attributes

protogenes, 2021-05-11:
By a quick look at `org.eclipse.milo.opcua.stack.core.AttributeId` milo is missing the following five attribute declarations (part 6 A.1 of the specification)
* DataTypeDefinition(23)
* RolePermissions(24)
* UserRolePermissions(25)
* AccessRestrictions(26)
* AccessLevelEx(27)

I require support for the DataTypeDefinition attribute on DataTypeNode for the simple definitions of custum structure types.
Not using a DataTypeDictionary is preferred in OPC UA version 1.04 (see part 5 D.1 of the specification)

Is support for this scheduled anytime soon? I may create a pull request for DataTypeDefinition if you can give me some pointers if there is anything to do besides extending AttributeId and DataTypeNode classes.

protogenes, 2021-05-11:
*DISCLAIMER* I have no deeper knowledge and understanding of the standards and no knowledge of the changes between 1.3 and 1.04.
Anyway, I thought you don't know until you try and gave it a shot https://github.com/eclipse/milo/compare/master...protogenes:issues/gh836

Adding support for DataTypeDefinition attribute was straightforward.
There are a few loose ends

1.  `org.eclipse.milo.opcua.sdk.client.nodes.UaDataTypeNode#setAttributeValue()` requires a context to decode the ExtensionObject for DataTypeDefinition (or maybe in UaNode#refresh)
2. `DataTypeNodeLoader` needs to provide the definitions, but I won't touch a generated file
3. `DataTypeDictionaryManager` may set the attribute when registering descriptions

As said, I have no idea about further implications of supporting that attribute or if it is an isolated feature of the specification and if it is feasible to include it before targeting OPC UA 1.04. It would just help me solve my problem at hand without moving away from milo (for now)

I'm looking forward to your thoughts.

P.S. I think it would be nice to include the targeted OPC UA version in the README.

protogenes, 2021-05-12:
Fair enough, I wouldn't bring incomplete support into my library as well.

Thank you for the list of required changes, after some discussion we will try to maintain a fork with DTD support for the meantime.

closed 2021-05-12 by protogenes

kevinherron, 2021-05-12:
FWIW we also need 1.04 support and the DataTypeDefinition stuff in particular where I work, so there's going to be more driving getting this done soon-ish than just my available spare time and motivation.