eclipse-milo/milo #842: Bad_SequenceNumberInvalid when closing connection

hutcheb, 2021-05-20:
**Describe the bug**
I have a test that sends/receives the following sequence of packets. The sequence number and time difference between messages is shown.

PublishRequest - Seq 6
PublishResponse - Seq 6 (td 0.95s)
CloseSessionRequest - Seq 8 (td 1.05s)
PublishRequest - Seq 7 (td 0.0003s)

Once the server receives the CloseSessionRequest it generates a Bad_SequenceNumberInvalid error.  (The server  doesn't seem to remove the monitored items, close the session or respond to further requests. but that looks like a separate issue)

I have been trying to setup a client which has multiple threads, in this case one for closing a session and another for the subscription. I have setup an ordered queue so that I shouldn't be sending messages out of order but this one seems to slip through for some reason. I'll probably work around it by delaying the CloseSessionRequest message for compatibility with other servers but I think the issue is still valid.

**Expected behavior**
From my understanding there is a requirement for messages to be ordered using the sequence number but there doesn't seem to be a requirement for messages to be received by the server or client in this order. I would expect that the CloseSessionRequest (Seq 8) be buffered by the server until it receives the PublishRequest (Seq 7).

In this case the client should be doing a better job at ordering the packets but I don't think the server shouldn't be relying on receiving them in the correct order. 

**Logs and Packet Captures**
[image]

**Additional context**
I am using the Milo Example Server 0.54, along with a development PLC4X branch (https://github.com/apache/plc4x/tree/feature/native_opua_client).

kevinherron, 2021-05-20:
Can you upload the logs and that Wireshark capture?

kevinherron, 2021-05-20:
If you are talking about the "Security Sequence Number" in the capture details then I'm afraid you have misunderstood the spec.

It is a hard requirement that message chunks are monotonically sequenced and sent/received by the client/server in order. Part 6 is quite clear that is required. It is used to prevent and detect replay detects and is treated as a security mechanism in addition to ensuring that message chunks arrive in order and can be properly decoded.

From _6.7.2.4 Sequence Header_
> The SequenceNumber shall also monotonically increase for all Messages and shall not wrap around until it is greater than 4 294 966 271 (UInt32.MaxValue – 1 024). The first number after the wrap around shall be less than 1 024. Note that this requirement means that a SequenceNumber does not reset when a new TokenId is issued. **The SequenceNumber shall be incremented by exactly one for each MessageChunk sent unless the communication channel was interrupted and re-established.** Gaps are permitted between the SequenceNumber for the last MessageChunk received before the interruption and the SequenceNumber for first MessageChunk received after communication was re-established. Note that the first MessageChunk after a network interruption is always an OpenSecureChannel request or response. **If gaps occur in any other case the receiver shall close the SecureChannel.**

From _6.7.6 Verifying Message Security_
> After the security validation is complete the receiver shall verify the RequestId and the SequenceNumber. If these checks fail a Bad_SecurityChecksFailed error is reported. The RequestId only needs to be verified by the Client since only the Client knows if it is valid or not.

The only thing Milo is doing incorrectly in this case is "leaking" `Bad_SequenceNumberInvalid` instead of using `Bad_SecurityChecksFailed`.

hutcheb, 2021-05-20:
Yeah you're right,

Another section to support you, in 6.7.1,
UASC requires a TransportProtocol that will preserve the order of MessageChunks, however, a UASC implementation does not necessarily process the Messages in the order that they were received.

Thanks

closed 2021-05-20 by hutcheb

kevinherron, 2021-05-20:
@hutcheb I'm curious why you guys are replacing the Milo-based client implementation for OPC UA with a homegrown one?

hutcheb, 2021-05-20:
I'm doing it because it is interesting, and I learn more about some of the protocols I use on a regular basis.

For PLC4X, the aim is to have a native implementation across the languages it supports Java/Go/C/... some of the code generation stuff in PLC4X allows it to be ported somewhat to the other languages automatically.

Although I'm kind of regretting starting with the OPC UA Java implementation as the original worked well and it's tough to beat ;)

kevinherron, 2021-05-20:
> I'm doing it because it is interesting, and I learn more about some of the protocols I use on a regular basis.

Heh, totally understand this :)

I just wanted to make sure you weren't experiencing some horrible bugs or instability with Milo that nobody had bothered reporting to me. The Milo client has been deployed in production in multiple products for a number of years now so I expect it to be pretty stable.

hutcheb, 2021-05-20:
No Milo has been flawless, all of the issues I’ve had with it have been
traced back to something stupid I’ve done :)