eclipse-milo/milo #845: ServiceFault when Deleting Subscriptions

hutcheb, 2021-05-24:
**Describe the bug**
Prior to closing a session, it is recommended to wait for any outstanding requests to be returned.
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.6.4/

When subscriptions are open we would have to wait for any PublishRequests to be returned which could be a long time depending on the publishinterval/keep alive.  Sending a DeleteSubscriptionsRequest should force the server to return any outstanding PublishRequests with a status code of Bad_NoSubscription.
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.13.8/

**Expected behavior**
There's isn't much description in the spec on what kind of message should be returned for this, at the moment a ServiceFault is returned. As it isn't a fault I would have expected just a PublishResponse with a ServiceResult of Bad_NoSubscription be returned.

**Logs and Packet Captures**
[image]

**Additional context**
Milo Server 0.5.4, PLC4X development branch client

kevinherron, 2021-05-24:
The spec says `ServiceFault` is used any time a service-level error occurs, which I've interpreted as any time the service-level StatusCode being returned is non-good.

hutcheb, 2021-05-24:
One argument against this would be that it would make the ServiceResult redundant.
I might check other servers just to document it.

kevinherron, 2021-05-24:
> One argument against this would be that it would make the ServiceResult redundant.

Sure, but the flip side to that argument is that if you can stick a Bad StatusCode into any response that makes ServiceFault redundant.

hutcheb, 2021-05-25:
I checked four different servers Kepware, open62541, Prosys and python-opcua.

Kepware returns a ServiceFault,
open62541 returns a PublishResponse,
Prosys returns a PublishResponse.
python-opcua doesn't return either.

Doesn't seem to have a standard implementation.

closed 2021-05-25 by hutcheb

kevinherron, 2021-05-25:
I’ll make an issue to see if we can get this clarified in the spec.

kevinherron, 2021-05-28:
@hutcheb Just FYI, I did open an issue for this in the OPC Foundation tracker and the spec is going to be clarified and the Compliance Test Tool updated eventually. Any Bad service result should result in a ServiceFault being returned, not the corresponding response type.

hutcheb, 2021-05-29:
Hey that's cool, good to get it clarified, thanks for that. :)