eclipse-milo/milo #846: Server not returned outstanding requests on CloseSession

hutcheb, 2021-05-24:
**Describe the bug**
When a CloseSessionRequest is sent and the DeleteSubscriptions flag is set, any outstanding subscription PublishRequests aren't returned.

**Expected behavior**
According to https://reference.opcfoundation.org/v104/Core/docs/Part4/5.6.4/
all outstandnig requests should be returned with a Bad_SessionClosed status.

**Logs and Packet Captures**
[image]

**Additional context**
Milo 0.5.4, PLC4X development branch.

kevinherron, 2021-05-24:
I think you are right that the server isn't sending Bad_SessionClosed for outstanding PublishRequests, but your test and capture don't correctly demonstrate this because you've already deleted the subscription before closing the Session and the only outstanding PublishRequest was returned as the ServiceFault with Bad_NoSubscription.

hutcheb, 2021-05-24:
[image]

Sorry had issues with the upload this morning

PR #857 (2021-06-07): Return Bad_SessionClosed for remaining PublishRequests
When the Session is closed and the delete subscriptions flag is set de-queue
any remaining PublishRequests and return a ServiceFault with Bad_SessionClosed.

fixes #846
commit 22656385 (2021-06-07): Return Bad_SessionClosed for remaining PublishRequests (#857)

When the Session is closed and the delete subscriptions flag is set de-queue
any remaining PublishRequests and return a ServiceFault with Bad_SessionClosed.

fixes #846
closed 2021-06-07 by kevinherron