eclipse-milo/milo #849: PluginExecutionException

Nefaden, 2021-05-25:
**Describe the bug**
While attempting to install dependencies, facing an execution error in stack-core dependency :

**Expected behavior**
Install depency simply and complete

**Logs and Packet Captures**
```
[INFO] stack-core ......................................... FAILURE [ 20.105 s]
```

```
[ERROR] Failed to execute goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest (generate-manifest) on project stack-core: Execution generate-manifest of goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest failed.: ConcurrentModificationException -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest (generate-manifest) on project stack-core: Execution generate-manifest of goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest failed.
```
I've visited the apache wiki for this issue, told me to report the bug

**Additional context**
What told me ```mvn -version```
```
Apache Maven 3.8.1 (05c21c65bdfed0f71a2f2ada8b84da59348c4c5d)
Maven home: /usr/local/Cellar/maven/3.8.1/libexec
Java version: 15.0.2, vendor: N/A, runtime: /usr/local/Cellar/openjdk/15.0.2/libexec/openjdk.jdk/Contents/Home
Default locale: en_FR, platform encoding: UTF-8
OS name: "mac os x", version: "11.3", arch: "x86_64", family: "mac"
```

I'm using VSCode  and iTerm2 for CLI

Hope we'll find a solution :)

kevinherron, 2021-05-25:
Can you try building with JDK 8 or 11 instead?

Nefaden, 2021-05-25:
Same errors with Java 8 or 11. Used these configurations : 
JDK8 
```
Apache Maven 3.8.1 (05c21c65bdfed0f71a2f2ada8b84da59348c4c5d)
Maven home: /usr/local/Cellar/maven/3.8.1/libexec
Java version: 1.8.0_292, vendor: AdoptOpenJDK, runtime: /Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home/jre
Default locale: en_FR, platform encoding: UTF-8
OS name: "mac os x", version: "10.16", arch: "x86_64", family: "mac"
```

JDK11
```
Apache Maven 3.8.1 (05c21c65bdfed0f71a2f2ada8b84da59348c4c5d)
Maven home: /usr/local/Cellar/maven/3.8.1/libexec
Java version: 11.0.11, vendor: AdoptOpenJDK, runtime: /Library/Java/JavaVirtualMachines/adoptopenjdk-11.jdk/Contents/Home
Default locale: en_FR, platform encoding: UTF-8
OS name: "mac os x", version: "11.3", arch: "x86_64", family: "mac"
```

I hope that was the good thing to do in order to execute maven based on jdk 8 or 11 but the fact is, it result in the same error

kevinherron, 2021-05-25:
Hmm, I'm not sure what's wrong with `maven-bundle-plugin`. 

I have a very similar setup and no issue building via `mvn clean install`:

```
kevin@kevins-mbp ~/D/milo (master)> mvn -version
Apache Maven 3.8.1 (05c21c65bdfed0f71a2f2ada8b84da59348c4c5d)
Maven home: /usr/local/Cellar/maven/3.8.1/libexec
Java version: 11.0.10, vendor: AdoptOpenJDK, runtime: /Library/Java/JavaVirtualMachines/adoptopenjdk-11.jdk/Contents/Home
Default locale: en_US, platform encoding: UTF-8
OS name: "mac os x", version: "10.16", arch: "x86_64", family: "mac"
```

Nefaden, 2021-05-25:
Ah maybe the 11.3 make the error ? Even if you're in 11.3 too but maven seems to see that you're under the 10.16 version. Before doing a force update in CLI, each IDE was under the 10.X version, never updated to BigSure.

I'll try with a full java IDE like IntelliJ but I think it'll not solved the problem.

kevinherron, 2021-05-25:
Searching for others with the same issue the common thread seems to be that this happens when building with JDK 15.

You might try updating the bundle plugin version to 4.2.1 and making sure that both `mvn -version` and `java -version` point to JDK 8 or 11 before attempting to build.

kevinherron, 2021-05-25:
Indeed, I can reproduced this building with JDK 15 but not 8 or 11:

```
kevin@kevins-mbp ~/D/milo (master)> mvn -version
Apache Maven 3.8.1 (05c21c65bdfed0f71a2f2ada8b84da59348c4c5d)
Maven home: /usr/local/Cellar/maven/3.8.1/libexec
Java version: 15.0.2, vendor: AdoptOpenJDK, runtime: /Users/kevin/Library/Java/JavaVirtualMachines/adopt-openjdk-15.0.2/Contents/Home
Default locale: en_US, platform encoding: UTF-8
OS name: "mac os x", version: "10.16", arch: "x86_64", family: "mac"
kevin@kevins-mbp ~/D/milo (master)> java -version
openjdk version "15.0.2" 2021-01-19
OpenJDK Runtime Environment AdoptOpenJDK (build 15.0.2+7)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 15.0.2+7, mixed mode, sharing)
```

```
[ERROR] Failed to execute goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest (generate-manifest) on project stack-core: Execution generate-manifest of goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest failed.: ConcurrentModificationException -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginExecutionException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :stack-core
```

Nefaden, 2021-05-25:
It seems so :
```
java -version
```
Does this : 
```
openjdk version "11.0.11" 2021-04-20
OpenJDK Runtime Environment AdoptOpenJDK-11.0.11+9 (build 11.0.11+9)
OpenJDK 64-Bit Server VM AdoptOpenJDK-11.0.11+9 (build 11.0.11+9, mixed mode)
```
And, for maven : 
```
Apache Maven 3.8.1 (05c21c65bdfed0f71a2f2ada8b84da59348c4c5d)
Maven home: /usr/local/Cellar/maven/3.8.1/libexec
Java version: 11.0.11, vendor: AdoptOpenJDK, runtime: /Library/Java/JavaVirtualMachines/adoptopenjdk-11.jdk/Contents/Home
Default locale: en_FR, platform encoding: UTF-8
OS name: "mac os x", version: "11.3", arch: "x86_64", family: "mac"
```

It lead to the same error : 
```
[ERROR] Failed to execute goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest (generate-manifest) on project stack-core: Execution generate-manifest of goal org.apache.felix:maven-bundle-plugin:4.0.0:manifest failed.: ConcurrentModificationException -> [Help 1]
```

kevinherron, 2021-05-25:
The latest version of `maven-bundle-plugin` seems to be `5.1.2`. Other reports of similar issues appear to be resolved by updating to that version... can you try that out and let me know?

Nefaden, 2021-05-25:
Ok, i didnt know how to do this, so I've changed the version in pom.xml like this : 
```
<groupId>org.apache.felix</groupId>
<artifactId>maven-bundle-plugin</artifactId>
<version>4.0.0</version> --> <version>5.1.2</version>
```

And it was enough, the build is successful, 

Thank you very much, you saved me a lot of time !

PR #850 (2021-05-26): Update maven-bundle-plugin to version 5.1.2
This resolves a ConcurrentModificationException some users see when attempting a build.

fixes #849
commit db7bdee5 (2021-05-26): Update maven-bundle-plugin to version 5.1.2 (#850)

This resolves a ConcurrentModificationException some users see when attempting a build.

fixes #849
closed 2021-05-25 by Nefaden