eclipse-milo/milo #853: getServerStatus() throws cast exception

ReinerZufall159, 2021-06-02:
I am using following milo example: "ReadNodeExample", when i run the Code i got a classCastException. I am not testing against the milo test server, i test against a modular control version XC303.

UaException: status=Bad_UnexpectedError, message=java.util.concurrent.ExecutionException: java.lang.ClassCastException: org.eclipse.milo.opcua.sdk.client.nodes.UaVariableNode cannot be cast to org.eclipse.milo.opcua.sdk.client.model.nodes.variables.ServerStatusTypeNode
	at org.eclipse.milo.opcua.sdk.client.model.nodes.objects.ServerTypeNode.getServerStatusNode(ServerTypeNode.java:369)
	at org.eclipse.milo.opcua.sdk.client.model.nodes.objects.ServerTypeNode.getServerStatus(ServerTypeNode.java:322)
	at opcClient.TestPrint.serverType(TestPrint.java:189)
	at opcClient.TestPrint.main(TestPrint.java:147)
Caused by: java.util.concurrent.ExecutionException: java.lang.ClassCastException: org.eclipse.milo.opcua.sdk.client.nodes.UaVariableNode cannot be cast to org.eclipse.milo.opcua.sdk.client.model.nodes.variables.ServerStatusTypeNode
	at java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:357)
	at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1908)
	at org.eclipse.milo.opcua.sdk.client.model.nodes.objects.ServerTypeNode.getServerStatusNode(ServerTypeNode.java:367)
	... 3 more
Caused by: java.lang.ClassCastException: org.eclipse.milo.opcua.sdk.client.nodes.UaVariableNode cannot be cast to org.eclipse.milo.opcua.sdk.client.model.nodes.variables.ServerStatusTypeNode
	at org.eclipse.milo.opcua.sdk.client.model.nodes.objects.ServerTypeNode.lambda$getServerStatusNodeAsync$23(ServerTypeNode.java:376)
	at java.util.concurrent.CompletableFuture.uniApply(CompletableFuture.java:616)
	at java.util.concurrent.CompletableFuture$UniApply.tryFire(CompletableFuture.java:591)
	at java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:488)
	at java.util.concurrent.CompletableFuture.complete(CompletableFuture.java:1975)
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:309)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

i know from #147 it is because the server return a wrong datatype which one cannot be cast. Is there a work around for this?

kevinherron, 2021-06-02:
You will have to do something similar to the workaround suggested in #147 if you want to continue to use the strongly typed information model.

You could also read the values directly using the NodeIds of the Nodes without using UaNode objects like in ReadExample.java.

kevinherron, 2021-06-02:
Also if the server is really not reporting the correct type definition for nodes in its information model then this should be reported to the vendor of the server as well.

closed 2021-06-02 by kevinherron