eclipse-milo/milo #861: Variable values written via OPC Client don't trigger onDataItemsModified

jorec, 2021-06-09:
**Describe the bug**
Variable values written via OPC Client don't trigger onDataItemsModified. 

**Expected behavior**
When OPC Client change the value of the variable via Write, there should be an event handler that will announce the change: e.g., onDataItemsModified.

**Logs and Packet Captures**

**Additional context**
Milo version: 0.6.1
Prosys client to change variable values (Write mode).

kevinherron, 2021-06-10:
`onDataItemsModified` is only triggered when the ModifyMonitoredItem service is called.

Writes to the value of a variable go directly to that Node. If you are using ManagedAddressSpace and UaNode then you would install an AttributeFilter to “intercept” the writes to Nodes you are interested in.

josiporec-wf, 2021-06-21:
@kevinherron thanks for your answer. 
Please also if it's possible, can you tell me is it safe to use `ctx.getSession().isEmpty();` to evaluate if the update comes from the server-side? 

I'm triggering an external service to write the value in this case, and only when it's from the client-side.

for example: 
```
boolean isServerUpdate = ctx.getSession().isEmpty();
        if (isServerUpdate) {
            ctx.setAttribute(attributeId, value);
        } else {
            doSomethingElse(ctx, value);
        }
```

Thank you.

jorec, 2021-06-21:
@kevinherron thanks for your answer.
Please also if it's possible, can you tell me is it safe to use `ctx.getSession().isEmpty();` to evaluate if the update comes from the server-side?

I'm triggering an external service to write the value in this case, and only when it's from the client-side.

for example:

```
boolean isServerUpdate = ctx.getSession().isEmpty();
        if (isServerUpdate) {
            ctx.setAttribute(attributeId, value);
        } else {
            doSomethingElse(ctx, value);
        }

```
Thank you.

kevinherron, 2021-06-21:
Yes, if `getSession()` is empty then somehow the attribute change is "internal" to the server.

closed 2021-06-28 by kevinherron