eclipse-milo/milo #867: How to deal with java.io.IOException: Connection reset by peer?

rxue92, 2021-06-24:
**Describe the bug**
After the client connected to server (TCP), I added monitored items to subscribe nodes. Sometimes I can see java.io.IOException: Connection reset by peer thrown by UascClientMessageHandler in log but the subscriptionListener didn't receive any message.

**Expected behavior**
I assume when Connection reset by peer is caught, the underlying TCP connection between client and server is gone. If so, since there is no response from SubscriptionListener, how can I know my client should reconnect to the server?

My implementation of SubscriptionListener will log when onPublishFailure, onStatusChanged, onSubscriptionTransferFailed, and onNotificationDataLost methods are called, but no log written when  Connection reset by peer caught and afterward.

**Logs and Packet Captures**
I think the stack is logged by printStackTrace method instead of by a logger (slf4j in my case).
```
2021-06-23 05:54:34.936 ERROR 1 --- [y-event-loop-53] o.e.m.o.s.c.t.u.UascClientMessageHandler : [remote=/192.168.***.***:4840] Exception caught: Connection reset by peer

java.io.IOException: Connection reset by peer
	at sun.nio.ch.FileDispatcherImpl.read0(Native Method) ~[na:1.8.0_292]
	at sun.nio.ch.SocketDispatcher.read(SocketDispatcher.java:39) ~[na:1.8.0_292]
	at sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:223) ~[na:1.8.0_292]
	at sun.nio.ch.IOUtil.read(IOUtil.java:192) ~[na:1.8.0_292]
	at sun.nio.ch.SocketChannelImpl.read(SocketChannelImpl.java:379) ~[na:1.8.0_292]
	at io.netty.buffer.PooledUnsafeDirectByteBuf.setBytes(PooledUnsafeDirectByteBuf.java:288) ~[netty-buffer-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.buffer.AbstractByteBuf.writeBytes(AbstractByteBuf.java:1132) ~[netty-buffer-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.channel.socket.nio.NioSocketChannel.doReadBytes(NioSocketChannel.java:347) ~[netty-transport-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:148) ~[netty-transport-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:677) [netty-transport-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:612) [netty-transport-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:529) [netty-transport-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:491) [netty-transport-4.1.33.Final.jar!/:4.1.33.Final]
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:905) [netty-common-4.1.33.Final.jar!/:4.1.33.Final]
	at java.lang.Thread.run(Thread.java:748) [na:1.8.0_292]
```

**Additional context**
```
    <properties>
        <project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>
        <project.reporting.outputEncoding>UTF-8</project.reporting.outputEncoding>
        <java.version>1.8</java.version>
    </properties>

        <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>sdk-client</artifactId>
            <version>0.5.4</version>
        </dependency>
        <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>sdk-server</artifactId>
            <version>0.5.4</version>
        </dependency>
```
The program is packaged to a jar file and run in a docker container which uses mcr.microsoft.com/java/jre:8u212-zulu-alpine as base image.

kevinherron, 2021-06-24:
You don't need to do anything.

Reconnecting, resuming or establishing a new session, and transferring subscriptions if necessary is all handled by the client in the background.

All you need to do is listen for `onSubscriptionTransferFailed` and re-create any Subscription that you get in this callback. This means a new Session was established and the server was unable to transfer your old Subscription to the new session.

rxue92, 2021-06-24:
Thanks for the quick reply. That's really clear and solves my problem.

closed 2021-06-24 by rxue92