eclipse-milo/milo #868: TCP connection not closed after failing to create a session

FSangouard, 2021-07-05:
**Describe the bug**
When failing to create a session on the OPC UA server because of e.g. bad configuration of certificates, the underlying TCP connection is never closed.
Running netstat still shows the connection as ESTABLISHED to the server.

**Expected behavior**
The TCP connection should be closed.

**Logs and Packet Captures**
[nifi-app-debug.log](https://github.com/eclipse/milo/files/6765609/nifi-app-debug.log)

According to the code that runs when the Nifi controller service shuts down... :
```
public void shutdown() {
        try {
            if (opcClient != null) {
                getLogger().debug("Disconnecting from OPC server...");
                opcClient.disconnect().get(3, TimeUnit.SECONDS);
            }
        } catch (Exception e) {
            getLogger().warn(e.getMessage());
        }
    }
```
...the last 3 lines of these logs indicate that the disconnect raises an exception.

**Additional context**
We're using a Nifi OPC UA package that uses Milo 0.2.1. Maybe this issue has already been fixed in later versions. I saw that there were many changes since this version but I'm not sure which one would fix this, if any. Maybe [this commit](https://github.com/eclipse/milo/commit/9cbc3d5e97f6ed4f7a4badc2f017d9c13f58b336) ?

kevinherron, 2021-07-06:
I don't know specifically if this was fixed since then, but this does not appear to be an issue in current versions of the client SDK.

FSangouard, 2021-07-07:
Thank you for your answer. Is it ok if I leave this issue open for now until we can test with the latest version?

closed 2021-07-22 by kevinherron