eclipse-milo/milo #869: UaServiceFaultException: status=Bad_UnexpectedError, message=An unexpected error occurred.

beyondyinjl2, 2021-07-20:
**Describe the bug**
try {
             subscription = client.getSubscriptionManager().createSubscription(publishInterval).get();
         } catch (Exception e) {
             log.error("OPC UA failed to create subscriber, {}", e.getMessage());
         }

Create subscriber error: UaServiceFaultException: status=Bad_UnexpectedError, message=An unexpected error occurred.

**Additional context**
<dependency>
<groupId>org.eclipse.milo</groupId>
<artifactId>sdk-client</artifactId>
<version>0.5.2</version>
</dependency>

OPCUA Server is WinCC

kevinherron, 2021-07-20:
This error seems to be coming from the server.

Can you get a Wireshark capture? Does this happen with other clients? Have you tried restarting WinCC?

beyondyinjl2, 2021-07-21:
After subscribing to 200 devices, some subscriptions will report this error. After restarting WinCC, there will still be several errors.

kevinherron, 2021-07-21:
Sounds like you should be in contact with Siemens support. There's not much I can do to help you with this being an error from the server and the very little information you've provided.

closed 2021-07-30 by kevinherron