eclipse-milo/milo #873: InstantiationCallback:: includeOptionalNode not working for the child hierarchy

comtel2000, 2021-08-02:
**Describe the bug**
When loading NodeSet models into the server namespace `getNodeFactory().createNode(..)`, any variables or objects with ModellingRule Optional  are instantiated only for the root level, but not for the child hierarchy even with `InstantiationCallback::includeOptionalNode`=true.

**Expected behavior**
`NodeFactory.InstantiationCallback::includeOptionalNode` should work for the whole tree hierarchy
 
The cached `InstanceDeclarationHierarchy` need to call the InstantiationCallback too?

**Additional context**
Eclipse Milo 0.6.3
UANodeSet Parser 0.4.0 (https://github.com/digitalpetri/opc-ua-nodeset-parser)
PackML NodeSet  (https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04/PackML)

kevinherron, 2021-08-02:
Hmm, `InstanceDeclarationHierarchy` should contain _all_ declared nodes, optional or mandatory, and then the callback should be invoked as these are being processed.

I will need to look at this a bit to see what's going on - what type are you instantiating from the PackML model?

comtel2000, 2021-08-02:
I instanciate a PackMLBaseObjectType object.

```xml
  <UAObjectType NodeId="ns=1;i=6" BrowseName="1:PackMLBaseObjectType">
    <DisplayName>PackMLBaseObjectType</DisplayName>
```
```java
(UaObjectNode) getNodeFactory().createNode(new NodeId(namespaceIndex, "PLM-DYN"), PackMLBaseObjectType.TYPE_ID, new NodeFactory.InstantiationCallback() { ..
```

and it works for the TagID and PackMLVersion optionals, but the admin object optionals are missing 

[image]

comtel2000, 2021-08-02:
I would expect to see all `PackMLObjects/PLM-DYN/Admin` Optionals too:

[image]

comtel2000, 2021-08-02:
The relation between Admin and the Optional 'AccTimeSinceReset' is:

```xml
  <UAVariable NodeId="ns=1;i=252" BrowseName="1:AccTimeSinceReset" ParentNodeId="ns=1;i=5" DataType="Int32">
    <DisplayName>AccTimeSinceReset</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5</Reference>
```

I guess `AccTimeSinceReset `is instanciated but was not linked to `Admin`.
`IsForward="true"` works but `IsForward="false"` doesn't?

kevinherron, 2021-08-02:
I can reproduce this easy enough but this code is so complicated I can't figure out what's actually broken or where. I'm not sure if the InstanceDeclarationHierarchy is being built wrong/incomplete or of the logic for creating node instances from type definitions needs to recursively use the NodeFactory.

It's going to take a little time for me to re-read the spec and try to wrap my head around this :/

kevinherron, 2021-08-02:
Try this patch (untested for regressions, but it does seem to create all the child nodes):

```
Index: opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java b/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java
--- a/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java	(revision d484f18a756e29837a2a1305eee8e1c640f57d57)
+++ b/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java	(date 1627927563419)
@@ -19,6 +19,7 @@
 import org.eclipse.milo.opcua.sdk.server.nodes.UaNode;
 import org.eclipse.milo.opcua.stack.core.Identifiers;
 import org.eclipse.milo.opcua.stack.core.NamespaceTable;
+import org.eclipse.milo.opcua.stack.core.types.builtin.ExpandedNodeId;
 import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
 import org.eclipse.milo.opcua.stack.core.types.enumerated.NodeClass;
 
@@ -130,7 +131,14 @@
                             .forEach(r -> referenceTable
                                 .addReference(browsePath, r.getReferenceTypeId(), r.getTargetNodeId()));
 
-                        addModeledNodes(node.getNodeId(), browsePath);
+                        Optional<ExpandedNodeId> instanceDeclarationTypeDefinitionId = node.getReferences().stream()
+                            .filter(r -> Identifiers.HasTypeDefinition.equals(r.getReferenceTypeId()))
+                            .findFirst()
+                            .map(Reference::getTargetNodeId);
+
+                        instanceDeclarationTypeDefinitionId
+                            .flatMap(xni -> xni.toNodeId(namespaceTable))
+                            .ifPresent(id -> addModeledNodes(id, browsePath));
                     }
                 })
             );
```

kevinherron, 2021-08-02:
Oops, lost the method arguments. This one looks a little better.

```
Index: opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java b/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java
--- a/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java	(revision d484f18a756e29837a2a1305eee8e1c640f57d57)
+++ b/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/nodes/factories/InstanceDeclarationHierarchy.java	(date 1627928803260)
@@ -15,10 +15,12 @@
 import java.util.stream.Collectors;
 
 import org.eclipse.milo.opcua.sdk.core.Reference;
+import org.eclipse.milo.opcua.sdk.core.nodes.MethodNode;
 import org.eclipse.milo.opcua.sdk.server.api.AddressSpaceManager;
 import org.eclipse.milo.opcua.sdk.server.nodes.UaNode;
 import org.eclipse.milo.opcua.stack.core.Identifiers;
 import org.eclipse.milo.opcua.stack.core.NamespaceTable;
+import org.eclipse.milo.opcua.stack.core.types.builtin.ExpandedNodeId;
 import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
 import org.eclipse.milo.opcua.stack.core.types.enumerated.NodeClass;
 
@@ -130,7 +132,18 @@
                             .forEach(r -> referenceTable
                                 .addReference(browsePath, r.getReferenceTypeId(), r.getTargetNodeId()));
 
-                        addModeledNodes(node.getNodeId(), browsePath);
+                        if (node instanceof MethodNode) {
+                            addModeledNodes(node.getNodeId(), browsePath);
+                        } else {
+                            Optional<ExpandedNodeId> instanceDeclarationTypeDefinitionId = node.getReferences().stream()
+                                .filter(r -> Identifiers.HasTypeDefinition.equals(r.getReferenceTypeId()))
+                                .findFirst()
+                                .map(Reference::getTargetNodeId);
+
+                            instanceDeclarationTypeDefinitionId
+                                .flatMap(xni -> xni.toNodeId(namespaceTable))
+                                .ifPresent(id -> addModeledNodes(id, browsePath));
+                        }
                     }
                 })
             );
```

[image]

comtel2000, 2021-08-03:
thanks, that works for me 👍

After hours with try and error patches and without any success, your's solve it. 
I tested it for the deeper hierarchy and now all Optionals are visible and also pushed into the filter `NodeFactory.InstantiationCallback::includeOptionalNode`.

Maybe part of another significantly easier ticket (API improvement):
`NodeFactory.InstantiationCallback::includeOptionalNode` should also provide information of the node browser path to
identify instances in case of duplicate browser names at different tree levels.

PR #874 (2021-08-04): Recursively add instance declarations from type definitions
When building an InstanceDeclarationHierarchy, add not only the explicitly
referenced instance declarations but also any instance declarations found
following the Node's type definition, if it has one. This allows optional
nodes not present on the instance declaration to later be instantiated,
if desired, by a NodeFactory.

fixes #873
commit a031aa42 (2021-08-04): Recursively add instance declarations from type definitions (#874)

When building an InstanceDeclarationHierarchy, add not only the explicitly
referenced instance declarations but also any instance declarations found
following the Node's type definition, if it has one. This allows optional
nodes not present on the instance declaration to later be instantiated,
if desired, by a NodeFactory.

fixes #873
closed 2021-08-04 by kevinherron