eclipse-milo/milo #875: java.security.InvalidKeyException: Illegal key size While running client-examples/ReadExample

xamry, 2021-08-08:
I imported the milo-master maven project in eclipse on my Windows machine. There are no compilation errors. I have also built it with maven. Now when I try to run the ReadExample to begin with, it throws below error:

00:11:29,427 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback.groovy]
00:11:29,427 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
00:11:29,427 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [file:/D:/milo-code/milo-master/milo-examples/client-examples/target/classes/logback.xml]
00:11:29,428 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs multiple times on the classpath.
00:11:29,428 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [file:/D:/milo-code/milo-master/milo-examples/client-examples/target/classes/logback.xml]
00:11:29,428 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [file:/D:/milo-code/milo-master/milo-examples/server-examples/target/classes/logback.xml]
00:11:29,706 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
00:11:29,751 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.ConsoleAppender]
00:11:29,754 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [STDOUT]
00:11:29,856 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - This appender no longer admits a layout as a sub-component, set an encoder instead.
00:11:29,856 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - To ensure compatibility, wrapping your layout in LayoutWrappingEncoder.
00:11:29,856 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - See also http://logback.qos.ch/codes.html#layoutInsteadOfEncoder for details
00:11:29,857 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to INFO
00:11:29,857 |-INFO in ch.qos.logback.core.joran.action.AppenderRefAction - Attaching appender named [STDOUT] to Logger[ROOT]
00:11:29,858 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
00:11:29,859 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@46f5f779 - Registering current configuration as safe fallback point

00:11:30.684 [milo-nonce-util-secure-random] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 334ms.
00:11:30.691 [main] INFO  o.e.m.examples.server.ExampleServer - security dir: C:\Users\AMRESH~1.SIN\AppData\Local\Temp\server\security
00:11:30.691 [main] INFO  o.e.m.examples.server.ExampleServer - security pki dir: C:\Users\AMRESH~1.SIN\AppData\Local\Temp\server\security\pki
00:11:30.694 [main] INFO  o.e.m.examples.server.KeyStoreLoader - Loading KeyStore at C:\Users\AMRESH~1.SIN\AppData\Local\Temp\server\security\example-server.pfx
00:11:50.343 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Stack version: 0.6.4-SNAPSHOT
00:11:50.344 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Server SDK version: 0.6.4-SNAPSHOT
00:11:52.140 [main] INFO  o.e.m.o.s.s.n.OpcUaNamespace - Loaded 2004 nodes in 768ms.
00:11:52.362 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://***********:12686/milo to 0.0.0.0:12686 [None/None]
00:11:53.180 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://***********:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
00:11:53.181 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://***********:12686/milo/discovery to 0.0.0.0:12686 [None/None]
00:11:53.181 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://10.212.135.208:12686/milo to 0.0.0.0:12686 [None/None]
00:11:53.181 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://10.212.135.208:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
00:11:53.181 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://10.212.135.208:12686/milo/discovery to 0.0.0.0:12686 [None/None]
00:11:53.181 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.56.1:12686/milo to 0.0.0.0:12686 [None/None]
00:11:53.181 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.56.1:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.56.1:12686/milo/discovery to 0.0.0.0:12686 [None/None]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://***********.****************:12686/milo to 0.0.0.0:12686 [None/None]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://***********.****************:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://***********.****************:12686/milo/discovery to 0.0.0.0:12686 [None/None]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://****************:12686/milo to 0.0.0.0:12686 [None/None]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://****************:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://****************:12686/milo/discovery to 0.0.0.0:12686 [None/None]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.1.5:12686/milo to 0.0.0.0:12686 [None/None]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.1.5:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.1.5:12686/milo/discovery to 0.0.0.0:12686 [None/None]
00:11:53.182 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo to 0.0.0.0:12686 [None/None]
00:11:53.183 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
00:11:53.183 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo/discovery to 0.0.0.0:12686 [None/None]
00:11:53.183 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://***********:8443/milo to 0.0.0.0:8443 [None/None]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://***********:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://***********:8443/milo/discovery to 0.0.0.0:8443 [None/None]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://10.212.135.208:8443/milo to 0.0.0.0:8443 [None/None]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://10.212.135.208:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://10.212.135.208:8443/milo/discovery to 0.0.0.0:8443 [None/None]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://192.168.56.1:8443/milo to 0.0.0.0:8443 [None/None]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://192.168.56.1:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://192.168.56.1:8443/milo/discovery to 0.0.0.0:8443 [None/None]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://***********.****************:8443/milo to 0.0.0.0:8443 [None/None]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://***********.****************:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
00:11:53.185 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://***********.****************:8443/milo/discovery to 0.0.0.0:8443 [None/None]
[... cut ...]
Caused by: java.security.InvalidKeyException: Illegal key size
	at javax.crypto.Cipher.checkCryptoPerm(Cipher.java:1039)
	at javax.crypto.Cipher.implInit(Cipher.java:805)
	at javax.crypto.Cipher.chooseProvider(Cipher.java:864)
	at javax.crypto.Cipher.init(Cipher.java:1396)
	at javax.crypto.Cipher.init(Cipher.java:1327)
	at org.eclipse.milo.opcua.stack.core.channel.ChunkEncoder$SymmetricEncoder.initCipher(ChunkEncoder.java:460)
	... 10 common frames omitted

**Additional context**
Eclipse Milo Version: 0.6.3 
Windows 10
Java 8

kevinherron, 2021-08-08:
I'm guessing you are running a version of Java 8 that is < 8u162, which means you aren't allowed to use 256-bit encryption without installing the JCE unlimited crypto policy files.

Easiest thing to do would be to use 8u162+, which no longer requires this: https://bugs.openjdk.java.net/browse/JDK-8170157

xamry, 2021-08-08:
Thanks @kevinherron . This solved my issue.

closed 2021-08-08 by kevinherron