eclipse-milo/milo #880: Session getting kicked out due to inactivity even though there is  activity.

Raghav-NK, 2021-09-03:
Hello @kevinherron , I'm facing an issue with respect to OPC UA connector.
if could you please help me with this that would be great.

Below I've mentioned the details issue.

--------------------------------------------------------------------Issue---------------------------------------------------------
Issue details :
we have 669 tags to read and so created  5 batches (means 5 connectors).
(why 5 batches? because we are not able to read all tags in one connector).

when we run all together, connectors stop reading in 2 mins.

**Expected behavior**
The session should not be kicked out if there is an activity.

**Logs and Packet Captures**

StatusCode{name=Bad_Timeout, value=0x800A0000, quality=bad}

So I have overridden the onStatusChanged method whenever the session gets kicked out we are recreating the subscription.

but still, it keeps getting every 3 to 4 mins 

**Additional context**
Milo version  is  0.2.4

################

kevinherron, 2021-09-03:
What makes you think the Session is being "kicked out"? The `onStatusChanged` method would be invoked if a _Subscription_ timed out on the server for some reason, which should only happen if the Subscription lifetime expires.

Can you check the settings you create the Subscription with and verify there are actually PublishRequests being sent regularly?

Raghav-NK, 2021-09-03:
Thanks for the quick response @kevinherron.

Right now subscription timeout is configured approx 7 days.

 cfg.setSessionTimeout(uint(600000000));
 cfg.setRequestTimeout(uint(5000));

Regarding PublishRequests,  how can we check this?

do we have to override the onPublishFailure method?

kevinherron, 2021-09-03:
Wireshark would be one way to check, however if you continue to get data changes or keep alive notifications delivered to the callbacks then it is safe to assume PublishRequests are being sent.

Raghav-NK, 2021-09-03:
How can we implement keep alive notifications call back in java OPC UA client application?

We are now observing in prosys opc ua client that data is getting changed every 1 second.

kevinherron, 2021-09-03:
`SubscriptionListener::onKeepAlive` is called when a "KeepAlive" response for the subscription is received. These are only sent in the absence of data change notifications, though.

Raghav-NK, 2021-09-03:
Okay.

is it possible to read 600 tags in one OPC UA connector( java client) ?

if yes, in our case what could be the reasons that we are failing to read all tags?

even after implementing multiple services which are dividing the load  130 tags each.

is there any problem with the server or with our client applicataions? 

could please you share your thoughts on this?

kevinherron, 2021-09-03:
Yes, 600 tags can be read or subscribed at once, unless the server has some limitations.

I can't tell you why it's not working. Pay attention to the StatusCodes that the server returns when you either issue the read request or create the monitored items.

Raghav-NK, 2021-09-03:
Thanks @kevinherron .

will check the status codes and monitor the behavior

closed 2021-09-20 by kevinherron