eclipse-milo/milo #881: Browsing Kepware tags doesn't work

hazaart, 2021-09-08:
**Describe the bug**
When browsing tags in a clean installed Kepware OPC server, only the root node is returning an array of 3 values: Objects, Views and Types. All of the Kepware tags are located under 'Objects'. But the browse function does not return any child nodes.

`List<Node> nodes = client.getAddressSpace().browse(browseRoot).get();`

**Expected behavior**
Should be able to find child nodes. It is strange one of the biggest OPC Servers is not supported at the moment.

kevinherron, 2021-09-11:
What version of KSE are you using?

kevinherron, 2021-09-11:
I have no issues pointing `BrowseNodeExample` at a local Kepware 6.7 instance and seeing all the nodes. You'll have to go into more detail about what you're doing.