eclipse-milo/milo #900: OPC client not quick reconnect when  kepware server restart

masterchengsheng, 2021-12-26:
**Describe the bug**
when I try restart kepware server ,I find client always throws 
org.eclipse.milo.opcua.stack.core.UaException: requestId=288 timed out after 30000ms
client runing in linux  ,About 20 minutes later, the system prompts“org.eclipse.milo.opcua.stack.core.UaException: connection closed”，And then call “onSubscriptionTransferFailed”。 I did not encounter this problem when I tested in the windows environment. As long as the kepware restarts or the server restarts, the onSubscriptionTransferFailed will be executed immediately in the windows environment.

**Expected behavior**
A clear and concise description of what you expected to happen instead.

**Logs and Packet Captures**
If applicable, add application logs or Wireshark captures to help explain your problem.

```
        return OpcUaClient.create(
                endpointUrl,
                endpoints ->
                        endpoints.stream()
                                .filter(el -> el.getEndpointUrl().equals(endpointUrl))
                                .findFirst(),
                configBuilder ->
                        configBuilder
                                .setApplicationName(LocalizedText.english("eclipse milo opc-ua client"))
                                .setApplicationUri("urn:eclipse:milo:examples:client")
                                .setCertificate(loader.getClientCertificate())
                                .setKeyPair(loader.getClientKeyPair())
                                .setIdentityProvider(new UsernameProvider(username, password))
                                .setRequestTimeout(uint(10000))//超时时间
                                .setKeepAliveFailuresAllowed(UInteger.MAX)
//                                .setKeepAliveFailuresAllowed(uint(10))
                                .setKeepAliveInterval(uint(10_000))
                                .build()
        );
    }
```

```

    class InnerSubscriptionListener implements UaSubscriptionManager.SubscriptionListener {
        @Override
        public void onKeepAlive(UaSubscription subscription, DateTime publishTime) {
            UaSubscriptionManager.SubscriptionListener.super.onKeepAlive(subscription, publishTime);
            log.warn(">>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onKeepAlive:{}", publishTime);
        }

        @Override
        public void onStatusChanged(UaSubscription subscription, StatusCode status) {
            UaSubscriptionManager.SubscriptionListener.super.onStatusChanged(subscription, status);
            log.warn(">>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onStatusChanged:{}", status);
        }

        @Override
        public void onPublishFailure(UaException exception) {
            log.error(">>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onPublishFailure", exception);
            UaSubscriptionManager.SubscriptionListener.super.onPublishFailure(exception);
            boolean cancelled = client.getSession().isCancelled();
            log.info("-----------------------------------------------session isCancelled:{}", cancelled);
        }

        @Override
        public void onNotificationDataLost(UaSubscription subscription) {
            UaSubscriptionManager.SubscriptionListener.super.onNotificationDataLost(subscription);
            log.warn(">>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onNotificationDataLost:{}", subscription);
        }

        @Override
        public void onSubscriptionTransferFailed(UaSubscription subscription, StatusCode statusCode) {
            UaSubscriptionManager.SubscriptionListener.super.onSubscriptionTransferFailed(subscription, statusCode);
            log.warn(">>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onSubscriptionTransferFailed:{}", statusCode);
            createSubscription();
        }
    }
```

2021-12-26 11:05:05.613 ERROR milo-shared-thread-pool-2 ---[331] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onPublishFailure
org.eclipse.milo.opcua.stack.core.UaException: requestId=288 timed out after 30000ms
	at org.eclipse.milo.opcua.stack.client.transport.AbstractTransport.lambda$scheduleRequestTimeout$7(AbstractTransport.java:137)
	at io.netty.util.HashedWheelTimer$HashedWheelTimeout.expire(HashedWheelTimer.java:669)
	at io.netty.util.HashedWheelTimer$HashedWheelBucket.expireTimeouts(HashedWheelTimer.java:744)
	at io.netty.util.HashedWheelTimer$Worker.run(HashedWheelTimer.java:469)
	at java.lang.Thread.run(Thread.java:748)
2021-12-26 11:05:05.614  INFO milo-shared-thread-pool-2 ---[334] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : -----------------------------------------------session isCancelled:false

**Additional context**
Version of Eclipse Milo you are using, OPC UA client or software server you are interacting with, etc...
OPC UA <version>0.6.3</version> 
OPC Server kepware 6.10

masterchengsheng, 2021-12-26:
org.eclipse.milo.opcua.stack.core.UaException: connection closed
	at org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler.channelInactive(UascClientMessageHandler.java:154)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:262)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:248)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:241)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelInactive(DefaultChannelPipeline.java:1405)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:262)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:248)
	at io.netty.channel.DefaultChannelPipeline.fireChannelInactive(DefaultChannelPipeline.java:901)
	at io.netty.channel.AbstractChannel$AbstractUnsafe$8.run(AbstractChannel.java:831)
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:164)
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:472)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:497)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.lang.Thread.run(Thread.java:748)

2021-12-26 11:05:23.866  INFO milo-shared-thread-pool-2 ---[334] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : -----------------------------------------------session isCancelled:false
2021-12-26 11:05:23.871 ERROR milo-shared-thread-pool-2 ---[331] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onPublishFailure

org.eclipse.milo.opcua.stack.core.UaException: connection closed
	at org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler.channelInactive(UascClientMessageHandler.java:154)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:262)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:248)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:241)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelInactive(DefaultChannelPipeline.java:1405)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:262)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:248)
	at io.netty.channel.DefaultChannelPipeline.fireChannelInactive(DefaultChannelPipeline.java:901)
	at io.netty.channel.AbstractChannel$AbstractUnsafe$8.run(AbstractChannel.java:831)
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:164)
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:472)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:497)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.lang.Thread.run(Thread.java:748)

2021-12-26 11:05:23.871  INFO milo-shared-thread-pool-2 ---[334] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : -----------------------------------------------session isCancelled:false
2021-12-26 11:05:24.894 ERROR milo-shared-thread-pool-2 ---[331] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onPublishFailure

org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_SessionIdInvalid, description=The session id is not valid.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:328)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

2021-12-26 11:05:24.895  INFO milo-shared-thread-pool-2 ---[334] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : -----------------------------------------------session isCancelled:false
2021-12-26 11:05:24.896 ERROR milo-shared-thread-pool-2 ---[331] [milo-shared-thread-pool-2] com.aimsphm.retrospect.opc.Client        : >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onPublishFailure

org.eclipse.milo.opcua.stack.core.UaServiceFaultException: status=Bad_SessionIdInvalid, description=The session id is not valid.
	at org.eclipse.milo.opcua.stack.client.UaStackClient.lambda$deliverResponse$5(UaStackClient.java:328)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$InlineTask.run(ExecutionQueue.java:154)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:136)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

[... cut ...]
2021-12-26 11:05:25.919  WARN milo-shared-thread-pool-0 ---[42] [milo-shared-thread-pool-0] ntCertificateValidator$InsecureValidator : Skipping validation for certificate: CN=KEPServerEX/UA Server, O=Unknown, C=CN, DC=JPZS-OPC2.yxcf.com
2021-12-26 11:05:25.953  WARN milo-shared-thread-pool-0 ---[347] [milo-shared-thread-pool-0] com.aimsphm.retrospect.opc.Client        : >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>onSubscriptionTransferFailed:StatusCode{name=Bad_SubscriptionIdInvalid, value=0x80280000, quality=bad}

masterchengsheng, 2021-12-26:
After testing, it is found that after the Kepware server restarts and disconnects, it takes about 15 minutes for the Client to call onSubscriptionTransferFailed. I hope that when my Kepware restarts successfully, I can immediately subscribe to the data in the OPC normally. At the same time, I also tested that when kepware restarted, I immediately opened a new Client and could get the corresponding data immediately.

kevinherron, 2021-12-26:
Don’t do this:
```
                                .setKeepAliveFailuresAllowed(UInteger.MAX)
```
Linux and Windows TCP stack behave differently when it comes to notifying that a socket connection has been lost in an “unclean” manner. You are effectively disabling a mechanism that exists to speed up detection of a connection that has been lost.

masterchengsheng, 2021-12-26:
I set the parameter to 100 and tested it, and it turned out that it was 15 minutes，Can I not set this parameter and let him default to 1
`configBuilder
                                .setApplicationName(LocalizedText.english("eclipse milo opc-ua client"))
                                .setApplicationUri("urn:eclipse:milo:examples:client")
                                .setCertificate(loader.getClientCertificate())
                                .setKeyPair(loader.getClientKeyPair())
                                .setIdentityProvider(new UsernameProvider(username, password))
                                .setRequestTimeout(uint(10000))//超时时间
//                                .setRequestTimeout(uint(5000))//超时时间
//                                .setKeepAliveFailuresAllowed(UInteger.MAX)
                                .setKeepAliveFailuresAllowed(uint(100))
                                .setKeepAliveInterval(uint(10_000))
                                .build()`

masterchengsheng, 2021-12-26:
I set the parameter to 100 and tested it, and it turned out that it was 15 minutes，Can I not set this parameter and let it default to 1   `                                .setKeepAliveFailuresAllowed(uint(100))
`

kevinherron, 2021-12-26:
You should probably leave all the keep alive settings at their default values.

closed 2021-12-28 by kevinherron