eclipse-milo/milo #901: Can't set nodeCache expiration time

flobz, 2022-01-12:
I would like to change the cache expiration paramater of the nodeCache but it seems impossible.
The constructor set expireAfterWrite to 2 minutes and it can't be changed:
```

public NodeCache() {
        this(b -> {});
    }
    public NodeCache(Consumer<CacheBuilder<Object, Object>> consumer) {
        CacheBuilder<Object, Object> builder = CacheBuilder.newBuilder()
            .expireAfterWrite(2, TimeUnit.MINUTES)
            .maximumSize(16384)
            .recordStats();

        consumer.accept(builder);

        this.cachedNodes = builder.build();
        this.canonicalNodes = new ConcurrentHashMap<>();
    }
```
I can propose a fix you like.

kevinherron, 2022-01-12:
Why can't it be changed? Does `CacheBuilder` not allow `expireAfterWrite` to be called a subsequent time?

flobz, 2022-01-12:
Yes exactly.

kevinherron, 2022-01-12:
Okay, I see an easy fix that doesn’t break any backwards compatibility.

kevinherron, 2022-01-12:
See https://github.com/eclipse/milo/pull/902, any objections?

flobz, 2022-01-12:
No it's perfect! :)

closed 2022-01-12 by kevinherron

flobz, 2022-01-12:
Thanks for the quick fix!

kevinherron, 2022-01-12:
Sure, no problem. I'll have think about actually getting this release out soon, now.

PR #902 (2022-01-12): Don't modify CacheBuilder in the constructor that accepts Consumer
fixes #901
commit c83f2939 (2022-01-12): Don't modify CacheBuilder in the constructor that accepts Consumer (#902)

fixes #901