eclipse-milo/milo #908: Factory doesn't create Method-Nodes

jnsw, 2022-01-25:
**Describe the bug**
I create Type-Nodes and want to make Object-Nodes from it. It works for Object- and Property-Nodes. But now I made some Method-Nodes, they can be seen in Types but in Objects they won't be shown in their place.

Using Milo 0.6.1.

kevinherron, 2022-01-25:
What kind of Reference did you use? Was there a Modeling Rule added as well? Can you provide an example?

It’s also possible this is the same issue as #874 and is fixed and pending release.

jnsw, 2022-01-26:
Thanks, maybe it's because of the modeling rule, I forgot to add. I close this, if this was the problem.

jnsw, 2022-01-26:
I am creating the method like in the example server but adding a reference "HasComponent" to a Node in BaseObjectTypes, which I later want to instantia in the Objects-branch with the "getNodeFactory().createNode(...)" - with the modelling rule added now, I get the method there, but it's still missing the input- and output-arguments in the objects-branch and says "not implemented"

jnsw, 2022-01-26:
In ObjectTypes in UaExpert you can see this with the OutputArguments as only Property, which is correct - I also can call this method here
[image]
I added the ModellingRule to this Method-Node and now it appears correctly in Objects, but without any argument-properties and I can't call it stating not implemented 

do I have to add modelling rules to its properties manually?

kevinherron, 2022-01-27:
The modeling rule must be present on the Method Node and both its InputArguments and OutputArguments Property Nodes if applicable.

Example from another model:
[image]
[image]

Note that both the "RemoteCommand" Method Node and its "InputArguments" Property Node have a modeling rule.

jnsw, 2022-01-28:
I only used the "setInputArguments" and "setOutputArguments" from your example, where you add them to methodNode in line 595 and 596 in ExampleNamespace. How can I get the corresponding argument-nodes to add my modeling rule to?

kevinherron, 2022-01-28:
You could call `getPropertyNode` or look it up by NodeId.

The example doesn't set these because it's building a method _instance_. You are trying to build a custom type and any of the Nodes hanging off a custom type are what the spec calls an _instance declaration_, and one of the things they need is a modeling rule reference.

jnsw, 2022-01-31:
With getPropertyNode I only get an PropertyNode not an PropertyTypeNode where it has the method "addReference" so I decided to use following, which puts the modeling rule to all two properties "InputArguments" and "OutputArguments", but it still doesn't show up in the instance tree.

```
        List<Node> argumentProperties = method.getPropertyNodes();
        for (Node property : properties){
            PropertyTypeNode propertyAsPTN = (PropertyTypeNode) property;
            propertyAsPTN.addReference(new Reference(propertyAsPTN.getNodeId(), Identifiers.HasModellingRule, Identifiers.ModellingRule_Mandatory.expanded(), true));
        }
```

jnsw, 2022-01-31:
I can't add it to ObjectTypeManager because`UaMethodNode::new` throws an error

```
        getServer().getObjectTypeManager().registerObjectType(
                methodNode.getNodeId(),
                UaMethodNode.class,
                UaMethodNode::new
        );
```

jnsw, 2022-01-31:
this doesn't work either, is doesn't throw errors but there is no modelling rule in the type tree
```
//        method.setOutputArguments(methodClass.getOutputArguments());
        method.setProperty(MethodNodeProperties.OutputArguments, methodClass.getOutputArguments());
        VariableNode outputProperty = method.getPropertyNode(MethodNodeProperties.OutputArguments).orElse(null);
        PropertyTypeNode outputProp2 = (PropertyTypeNode) outputProperty;
        outputProp2.addReference(new Reference(outputProp2.getNodeId(),Identifiers.HasModellingRule, Identifiers.ModellingRule_Mandatory.expanded(), true));
```