eclipse-milo/milo #912: Connecting to OPC UA Server with UAExpert or ProsysOPCUABrowser via opc.https

markuzzi, 2022-02-01:
Hi, first of all, thanks very much for the library!

I am trying to expose the OPC UA server via the HTTPS endpoint.
However, I am not able to connect to the server with UAExpert of the Prosys OPC UA Browser.

The server log says: io.netty.handler.ssl.NotSslRecordException: not an SSL/TLS record.
And indeed, if you look into the payload with xxd (hex editor), it is a standard HTTP request without SSL.

It's actually very easy to reproduce. Just run the example milo server and try to connect with any client to the HTTPS endpoint.

Any idea, what am I missing?

Thanks for your help,
Markus

kevinherron, 2022-02-01:
The HTTPS transport is not fully implemented or tested. When I stopped working on it only the HTTPS + Binary encoding was possibly working and both of those clients only supported HTTPS + XML encoding. I don’t know if that has changed, but it’s best to stick with the OPC.TCP transport instead.

kevinherron, 2022-02-01:
Hmm, I am able to successfully connect UaExpert with HTTPS + UA Binary encoding.

There are some warnings in the UaExpert logs but the connection seems to work.

closed 2022-02-05 by kevinherron