eclipse-milo/milo #917: Parsing NodeSet xml file (incl. data/object types, aliases, extensions) and integrating into OPC UA server

WachTy, 2022-02-09:
Hi,

I´m using [opc-ua-nodeset-parser](https://github.com/digitalpetri/opc-ua-nodeset-parser) to map an already exisiting data structure (exported as NodeSet xml file) as NodeSet object and integrate it into my milo OPC UA server.

Creating UaObjects, UaVariables, UaObjectTypes and UaDataTypes in my server workes fine, but I miss the possibility to create aliases (as available for [.NET SDK bundle](https://documentation.unified-automation.com/uasdknet/2.2.2/html/classUnifiedAutomation_1_1UaSchema_1_1UANodeSet.html#ad197cd146436d65d257ad7fea74c5901)) and extensions and add them to the server (like nodes).
Is it possible using the SDK to map a Nodeset xml file (incl. aliases & extensions) into a OPC UA server?

kevinherron, 2022-02-09:
There's no official support for what you doing at all right now. That nodeset-parser project is in take it or leave it condition.

That said, I'm not clear on whether you're talking about the alias functionality in the UANodeSet schema, which just exists to increase readability by substation often used or well known NodeIds with their Names inside the document, or if you're talking about something else.

kevinherron, 2022-02-09:
The parser already resolves aliases into NodeIds as it creates the attribute objects that result from the parse. You should not need to do anything with the alias table.

One example: https://github.com/digitalpetri/opc-ua-nodeset-parser/blob/76ff6ce459b902a6fe2d2ab154ec8fcdf2064330/src/main/java/com/digitalpetri/opcua/nodeset/UaNodeSet.java#L101

`tryParseNodeId` receives the alias table and does a lookup if necessary.

closed 2022-02-11 by kevinherron