eclipse-milo/milo #920: Error creating OpcUaClient. UnsupportedOperationException: Reflective setAccessible(true) disabled

MasterLi, 2022-02-09:
Error creating OpcUaClient. SpringBoot Version=2.2.6.RELEASE
org.eclipse.milo:sdk-client version: 0.6.3
The server is not available at the specified address, the address is correct. If the server is available, then there is no error.

There should be no error when creating a client.
url=opc.tcp://10.40.254.01:4840

OpcUaClient.create(url,
                endpoints ->
                        endpoints.stream()
                                .map(endpoint -> updateEndpointUrl(endpoint, url))
                                .findFirst(),
                configBuilder -> configBuilder
                        .setApplicationName(LocalizedText.english("default"))
                        .setApplicationUri("default")
                        .setRequestTimeout(uint(5_000))
                        .setConnectTimeout(uint(5_000))
                        .build());

java.lang.UnsupportedOperationException: Reflective setAccessible(true) disabled
	at io.netty.util.internal.ReflectionUtil.trySetAccessible(ReflectionUtil.java:31)
	at io.netty.util.internal.PlatformDependent0$4.run(PlatformDependent0.java:225)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at io.netty.util.internal.PlatformDependent0.<clinit>(PlatformDependent0.java:219)
	at io.netty.util.internal.PlatformDependent.isAndroid(PlatformDependent.java:289)
	at io.netty.util.internal.PlatformDependent.<clinit>(PlatformDependent.java:92)
	at io.netty.channel.nio.NioEventLoop.newTaskQueue0(NioEventLoop.java:279)
	at io.netty.channel.nio.NioEventLoop.newTaskQueue(NioEventLoop.java:150)
	at io.netty.channel.nio.NioEventLoop.<init>(NioEventLoop.java:138)
	at io.netty.channel.nio.NioEventLoopGroup.newChild(NioEventLoopGroup.java:146)
	at io.netty.channel.nio.NioEventLoopGroup.newChild(NioEventLoopGroup.java:37)
	at io.netty.util.concurrent.MultithreadEventExecutorGroup.<init>(MultithreadEventExecutorGroup.java:84)
	at io.netty.util.concurrent.MultithreadEventExecutorGroup.<init>(MultithreadEventExecutorGroup.java:58)
	at io.netty.util.concurrent.MultithreadEventExecutorGroup.<init>(MultithreadEventExecutorGroup.java:47)
	at io.netty.channel.MultithreadEventLoopGroup.<init>(MultithreadEventLoopGroup.java:59)
	at io.netty.channel.nio.NioEventLoopGroup.<init>(NioEventLoopGroup.java:86)
	at io.netty.channel.nio.NioEventLoopGroup.<init>(NioEventLoopGroup.java:81)
	at io.netty.channel.nio.NioEventLoopGroup.<init>(NioEventLoopGroup.java:68)
	at org.eclipse.milo.opcua.stack.core.Stack.sharedEventLoop(Stack.java:81)
	at org.eclipse.milo.opcua.stack.client.UaStackClientConfigBuilder.build(UaStackClientConfigBuilder.java:130)
	at org.eclipse.milo.opcua.stack.client.DiscoveryClient.getEndpoints(DiscoveryClient.java:212)
	at org.eclipse.milo.opcua.stack.client.DiscoveryClient.getEndpoints(DiscoveryClient.java:195)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.create(OpcUaClient.java:213)
	at ru.asuer.collector.client.opcua.OpcUaClientWrapper.createClient(OpcUaClientWrapper.java:169)
	at ru.asuer.collector.client.opcua.OpcUaClientWrapper.connect(OpcUaClientWrapper.java:80)

java.lang.IllegalAccessException: class io.netty.util.internal.PlatformDependent0$6 cannot access class jdk.internal.misc.Unsafe (in module java.base) because module java.base does not export jdk.internal.misc to unnamed module @3273a9d4
	at java.base/jdk.internal.reflect.Reflection.newIllegalAccessException(Reflection.java:361)
	at java.base/java.lang.reflect.AccessibleObject.checkAccess(AccessibleObject.java:591)
	at java.base/java.lang.reflect.Method.invoke(Method.java:558)
	at io.netty.util.internal.PlatformDependent0$6.run(PlatformDependent0.java:335)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at io.netty.util.internal.PlatformDependent0.<clinit>(PlatformDependent0.java:326)
	at io.netty.util.internal.PlatformDependent.isAndroid(PlatformDependent.java:289)
	at io.netty.util.internal.PlatformDependent.<clinit>(PlatformDependent.java:92)
	at io.netty.channel.nio.NioEventLoop.newTaskQueue0(NioEventLoop.java:279)
	at io.netty.channel.nio.NioEventLoop.newTaskQueue(NioEventLoop.java:150)
	at io.netty.channel.nio.NioEventLoop.<init>(NioEventLoop.java:138)
	at io.netty.channel.nio.NioEventLoopGroup.newChild(NioEventLoopGroup.java:146)
	at io.netty.channel.nio.NioEventLoopGroup.newChild(NioEventLoopGroup.java:37)
	at io.netty.util.concurrent.MultithreadEventExecutorGroup.<init>(MultithreadEventExecutorGroup.java:84)
	at io.netty.util.concurrent.MultithreadEventExecutorGroup.<init>(MultithreadEventExecutorGroup.java:58)
	at io.netty.util.concurrent.MultithreadEventExecutorGroup.<init>(MultithreadEventExecutorGroup.java:47)
	at io.netty.channel.MultithreadEventLoopGroup.<init>(MultithreadEventLoopGroup.java:59)
	at io.netty.channel.nio.NioEventLoopGroup.<init>(NioEventLoopGroup.java:86)
	at io.netty.channel.nio.NioEventLoopGroup.<init>(NioEventLoopGroup.java:81)
	at io.netty.channel.nio.NioEventLoopGroup.<init>(NioEventLoopGroup.java:68)
	at org.eclipse.milo.opcua.stack.core.Stack.sharedEventLoop(Stack.java:81)
	at org.eclipse.milo.opcua.stack.client.UaStackClientConfigBuilder.build(UaStackClientConfigBuilder.java:130)
	at org.eclipse.milo.opcua.stack.client.DiscoveryClient.getEndpoints(DiscoveryClient.java:212)
	at org.eclipse.milo.opcua.stack.client.DiscoveryClient.getEndpoints(DiscoveryClient.java:195)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.create(OpcUaClient.java:213)
	at ru.asuer.collector.client.opcua.OpcUaClientWrapper.createClient(OpcUaClientWrapper.java:169)

kevinherron, 2022-02-09:
Try this: https://stackoverflow.com/a/69726236/85927

MasterLi, 2022-02-10:
This hasn't been working since 11 java. You need to update the version of the library, this bug has been fixed in the new version. I temporarily fixed the issue as follows.
dependencies {
    implementation('io.netty:netty-common') {
        version {
            strictly '4.1.73.Final'
            prefer '4.1.73.Final'
        }
    }

closed 2022-02-11 by kevinherron