eclipse-milo/milo #923: Exception while registering Codec: no codec registered for encodingId

ThorsEnte, 2022-02-10:
So basically i have a similiar problem with working with the extensionObject. More properly with registering the CostumCodec. I followed the answer step by step in https://stackoverflow.com/questions/64935020/how-to-correctly-work-with-extensionobject-and-struct-in-milo-opc-ua but unfortunately i get an Exception while decoding the Object (Exception=org.eclipse.milo.opcua.stack.core.UaSerializationException: no codec registered for encodingId=NodeId{ns=4, id=3})  

My Code looks like in the Programm.txt (The code highlighting is broken, i cannot highlight it properly here)
I know the decoding function in class MftInCustomDataType isn't fully implemented yet, but while debugging i see it won't be called so far. Like the exception is telling me. 

According to the other post i tried to identify the NodeIds in UaExpert. The following pictures are from said Program.
[Should be the TYPE_ID][1]

And if i click on DefaultBinary i should get the Binary_id
[Should be the Binary_encoding_id][2]

And of course the actual node which i want to read
[Acutal node with ID][3]
I wonder why the exception tells me the encoding id is ns=4;id=3 an not the registered id from UaExpert.
I also tried this node Id for the binary Encoding type. [Also tried this node id for binary encoding][4]

Hopefully someone knows how to deal with this issue. (Server is from Siemens CPU 1510SP) Milo Version:0.6.3

  [1]: https://i.stack.imgur.com/xYirz.png
  [2]: https://i.stack.imgur.com/yPutD.png
  [3]: https://i.stack.imgur.com/3RjJy.png
  [4]: https://i.stack.imgur.com/cN9j3.png
[Programm.txt](https://github.com/eclipse/milo/files/8040810/Programm.txt)

Only if you don't like to download the .txt
{Code}
    public void test(OpcUaClient client) throws UaException {
        // synchronous read request via VariableNode
        registerCustomCodec(client);
        UaVariableNode node = client.getAddressSpace().getVariableNode(new NodeId(4, 4));
        logger.info("DataType={}", node.getDataType());

        // Read the current value
        DataValue value = node.readValue();
        logger.info("Value={}", value);

        Variant variant = value.getValue();
        ExtensionObject xo = (ExtensionObject) variant.getValue();
        try {
            SerializationContext tempContext = client.getStaticSerializationContext();
            xo.decode(tempContext);
        } catch (Exception e) {
            logger.info("Exception={}", e.toString());
        }
    }

    private void registerCustomCodec(OpcUaClient client) {
        //Register MfttoOpcuaDatatype
        MftInCustomDataType tempInMft = new MftInCustomDataType();
        NodeId dataTypeId = tempInMft.getTypeId()
                .toNodeId(client.getNamespaceTable())
                .orElseThrow(() -> new IllegalStateException("namespace not found"));

        NodeId binaryEncodingId = tempInMft.getBinaryEncodingId()
                .toNodeId(client.getNamespaceTable())
                .orElseThrow(() -> new IllegalStateException("namespace not found"));

        // Register codec with the client's DataTypeManager instance.
        // We need to register it by both its encodingId and its dataTypeId because it may be
        // looked up by either depending on the context.

        client.getStaticDataTypeManager().registerCodec(
                binaryEncodingId,
                new MftInCustomDataType.Codec().asBinaryCodec()
        );

        client.getStaticDataTypeManager().registerCodec(
                new QualifiedName(dataTypeId.getNamespaceIndex(), "MftInCustomDataType"),
                dataTypeId,
                new MftInCustomDataType.Codec().asBinaryCodec()
        );
    }

public class MftInCustomDataType implements UaStructure {
    private boolean baz;
    private ExpandedNodeId TYPE_ID = ExpandedNodeId.parse("ns=3;s=TD_\"OPC_UA_IN\"");
    private ExpandedNodeId BINARY_ENCODING_ID = ExpandedNodeId.parse("ns=3;s=TE_\"OPC_UA_IN\"");
    public MftInCustomDataType(boolean baz){
        this.baz = baz;
    }

    public MftInCustomDataType(){
    }

    @Override
    public ExpandedNodeId getTypeId() {
        return TYPE_ID;
    }

    @Override
    public ExpandedNodeId getBinaryEncodingId() {
        return BINARY_ENCODING_ID;
    }

    @Override
    public ExpandedNodeId getXmlEncodingId() {
        // XML encoding not supported
        return ExpandedNodeId.NULL_VALUE;
    }

    public static class Codec extends GenericDataTypeCodec<MftInCustomDataType> {
        @Override
        public Class<MftInCustomDataType> getType() {
            return MftInCustomDataType.class;
        }

        @Override
        public MftInCustomDataType decode(
                SerializationContext context,
                UaDecoder decoder) throws UaSerializationException {

            boolean foo = decoder.readBoolean("Tooglebyte");

            return new MftInCustomDataType(foo);
        }

        @Override
        public void encode(SerializationContext context, UaEncoder writer, MftInCustomDataType value) throws UaSerializationException {

        }
    }
}

{/Code}

kevinherron, 2022-02-10:
Is there a reason you're trying to hand-code structures and their definitions? The DataType info can be read and decoded automatically from Siemens PLC OPC UA servers. Depending on the firmware version you may have to enable "Legacy data type definitions", but once you've done that, you only need to register a `DataTypeDictionarySessionInitializer` and it will be read automatically.

See https://github.com/eclipse/milo/blob/master/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/UnifiedAutomationReadCustomDataTypeExample.java

ThorsEnte, 2022-02-10:
I was just following the example here and the points from the Question [[Exception while registering Codec: no codec registered for encodingId](https://stackoverflow.com/questions/71063214/exception-while-registering-codec-no-codec-registered-for-encodingid). 

But even with the UnifiedAutomationReadCustomDataTypeExample i get the same Exception description for Node (4,4).

But to be honest i'm not completely sure if i enabeld the legacy data type definitions right in TIA Portal. But i guess you mean under device settings->opc ua->server->backward compatibility data type defintion enable TypeDictionary activate. FYI firmware version is V2.8

closed 2022-02-11 by kevinherron

ThorsEnte, 2022-02-22:
Unfortunately i was sick the last few day so i couldn't work on this project till today, but i managed to get the log from DataTypeDictionaryReader
[opcua.log](https://github.com/eclipse/milo/files/8115407/opcua.log)
.

kevinherron, 2022-02-22:
> What's the value of DataType attribute in UaExpert? The screenshot cuts it off.

Can you get this info? Like I said, I don't think that datatypes you create via the "Service Interface" actually get a DataType Dictionary, so you may not be able to read them.

ThorsEnte, 2022-02-23:
Sure, hopefully it's the right one.
And also thanks for your help !

[image]

kevinherron, 2022-02-23:
See how the NodeId for that DataType is actually `ns=4;i=2`? I don’t think the server provides a DataType Dictionary for namespaces belonging to Service Interfaces.

You might be able to define the codec manually as you tried originally now that you know the DataType NodeId, assuming the corresponding is still `ns=4;i=3`, but I don’t know if this information is readily discoverable in the address space or not.