eclipse-milo/milo #931: [usability issue] OPCUaServerConfigBuilder has deprecated method without comment on what to do instead. Examples still use deprecated method.

JonasJurczok, 2022-02-25:
In Milo 0.6.4 there is an [ExampleServer](https://github.com/eclipse/milo/blob/master/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleServer.java#L168) that we use for test purposes.

In line 168 it uses the setHttpsCertificate.
In this [PR](https://github.com/eclipse/milo/pull/871/files) the method has been deprecated. However there is no comment on what to do instead. And the example server still uses the deprecated method.

To improve usability and lower the entrance barrier can you please
* add a comment on the deprecated method that describes how users of the API should react
* adapt the sample code to make it visible for people reading through the samples.

Thank you

JonasJurczok, 2022-02-25:
I created a PR to "solve" this. Feel free to use it however you like and/or your rules require you to :)

kevinherron, 2022-02-25:
Thanks, updated. Be careful using the HTTPS transport, it's not fully implemented or tested. At the time I was working on it I think only the HTTPS + Binary for a client was really being tested, and it was difficult to find other products that had implemented the combination.

JonasJurczok, 2022-02-25:
Thank you.

And thanks for the heads up. What is the recommended transport?

kevinherron, 2022-02-25:
The "opc.tcp" transport is the only transport required by all implementations and the only one I've ever seen used in the real world.

closed 2022-02-25 by kevinherron

JonasJurczok, 2022-02-25:
Thanks. We will switch to that then :)

PR #933 (2022-02-25): Explain setHttpsCertificate deprecation and update ExampleServer
fixes #931
commit ec070a4e (2022-02-25): Explain setHttpsCertificate deprecation and update ExampleServer (#933)

fixes #931