eclipse-milo/milo #942: After some time collecting data from a kepServer we get the error:  LEAK: ByteBuf.release() was not called before it's garbage-collected.

entaoG, 2022-03-12:
**Describe the bug**
After some time collecting data from a kepServer we get the error
**Expected behavior**

**Logs and Packet Captures**
After some time collecting data from a kepServer we get the error:

01:00:29.830 [milo-shared-thread-pool-1] ERROR i.n.u.ResourceLeakDetector - [reportTracedLeak,319] - LEAK: ByteBuf.release() was not called before it's garbage-collected. See https://netty.io/wiki/reference-counted-objects.html for more information.
Recent access records: 
Created at:
	io.netty.buffer.PooledByteBufAllocator.newDirectBuffer(PooledByteBufAllocator.java:385)
	io.netty.buffer.AbstractByteBufAllocator.directBuffer(AbstractByteBufAllocator.java:187)
	io.netty.buffer.AbstractByteBufAllocator.directBuffer(AbstractByteBufAllocator.java:178)
	io.netty.buffer.AbstractByteBufAllocator.ioBuffer(AbstractByteBufAllocator.java:139)
	io.netty.channel.DefaultMaxMessagesRecvByteBufAllocator$MaxMessageHandle.allocate(DefaultMaxMessagesRecvByteBufAllocator.java:114)
	io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:150)
	io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:719)
	io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:655)
	io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:581)
	io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)
	io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	java.lang.Thread.run(Thread.java:748)

**Additional context**
milo 0.6.1
KEPServerEX 6.6

kevinherron, 2022-03-12:
Is this repeatable? 

Can you set the JVM param `-Dio.netty.leakDetectionLevel=advanced` until it happens again and then get the full logs?

We have many customers connected to Kepware but I haven't seen any reports of this warning, so I'm curious to find out what is causing it.

entaoG, 2022-03-22:
Hi~
monitoring parameters （- Dio.net ty. LeakDetectionLevel=PARANOID ）added later, on March 15, 2022, was found in the log this kind of problem, but before and ([milo - Shared - thread pool - 1]),  And now it's [author-nioeventloop-4-1].  
**Logs：**
17:21:11.547 [lettuce-nioEventLoop-4-1] ERROR i.n.u.ResourceLeakDetector - [reportTracedLeak,319] - LEAK: ByteBuf.release() was not called before it's garbage-collected. See https://netty.io/wiki/reference-counted-objects.html for more information.
Recent access records: 
#1:
	io.netty.buffer.AdvancedLeakAwareByteBuf.readBytes(AdvancedLeakAwareByteBuf.java:496)
	com.ruoyi.microwave.netty.NettyClientHandler$1.call(NettyClientHandler.java:150)
	java.util.concurrent.FutureTask.run(FutureTask.java:266)
	java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	java.lang.Thread.run(Thread.java:748)
**Error location description reported**
Create a Netty client using Java, connect to the server, receive the server push message, locate the code location is (the third line below) : 
ByteBuf byteBuf = (ByteBuf) msg;
byte[] bytes = new byte[byteBuf.readableBytes()];
byteBuf.readBytes(bytes);

kevinherron, 2022-03-22:
Sorry I'm not sure what you're trying to tell me. The latest leak stack you provided is coming from `com.ruoyi.microwave.netty.NettyClientHandler`, which isn't part of Milo.

closed 2022-04-13 by kevinherron