eclipse-milo/milo #952: Adding value to DataTypeDefinition field of a DataType node

achintya-kumar, 2022-03-25:
Hi Kevin,

I am working with a custom nodeset specification XML file where there are DataType nodes as follows:
[image]

There are these elements called **Field** therein which have to be parsed. I'm currently using [opc-ua-nodeset-parser](https://github.com/digitalpetri/opc-ua-nodeset-parser) for parsing these XML files. 

I am looking for examples on how to create these fields and set the DataTypeDefinition attribute of the DataType node to look like this:
[image]

I have tried [this example](https://github.com/eclipse/milo/blob/master/milo-examples/server-examples/src/main/java/org/eclipse/milo/examples/server/ExampleNamespace.java#L824) from the **example-server** module but it doesn't seem to be generating anything on the related node's DataTypeDefinition attribute.

Any help would be very much appreciated.

Thank you!

kevinherron, 2022-03-25:
Unfortunately `DataTypeDefinition` is a new attribute that was introduced by OPC UA 1.04.

Milo is strictly OPC UA 1.03 right now. I don't think you're going to have any luck with this for now.

The good news is that I've finally started working on OPC UA 1.04 support for what will be Milo 2.0, the bad news is it's probably going to be a while before anything is ready.

achintya-kumar, 2022-03-25:
@kevinherron Thank you for your reply. All the best. Please let me know if I can also help.

closed 2022-03-25 by achintya-kumar