eclipse-milo/milo #955: how to get subscriberinfo

sangjiexun, 2022-04-02:
Dear developer, how can I get the subscriber (OPCUA Client) information from OPCUA Server, such as remote device's ip ,address and  computer-name, etc.

kevinherron, 2022-04-02:
If you're using the server SDK then the `Session` object has a `getClientAddress` method on it: https://github.com/eclipse/milo/blob/f5799292024a9fb8c9a025e0664c49d67c1110bf/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/Session.java#L251-L260

sangjiexun, 2022-04-02:
> If you're using the server SDK then the `Session` object has a `getClientAddress` method on it:
> 
> https://github.com/eclipse/milo/blob/f5799292024a9fb8c9a025e0664c49d67c1110bf/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/Session.java#L251-L260

THANKS!!!