eclipse-milo/milo #962: Java 17 module access violation

cloos-stone, 2022-04-14:
Hello together!

### Bug description
we recently upgraded our application to Java 17 and after some time we recognized that the access with e.g. UaExpert/Prosys was broken. When trying to read the value of a Node with a custom structure, I can't read the value. That worked in the past. 

### Logs
When I pack my application into a jar and run it, I get this exception:
```Apr 14, 2022 7:58:07 AM com.sun.xml.bind.v2.runtime.reflect.opt.Injector <clinit>
SEVERE: null
java.security.PrivilegedActionException: java.lang.NoSuchMethodException: sun.misc.Unsafe.defineClass(java.lang.String,[B,int,int,java.lang.ClassLoader,java.security.ProtectionDomain)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:573)
        at com.sun.xml.bind.v2.runtime.reflect.opt.Injector.<clinit>(Injector.java:166)
        at com.sun.xml.bind.v2.runtime.reflect.opt.AccessorInjector.prepare(AccessorInjector.java:51)
        at com.sun.xml.bind.v2.runtime.reflect.opt.OptimizedAccessorFactory.get(OptimizedAccessorFactory.java:157)
        at com.sun.xml.bind.v2.runtime.reflect.Accessor$FieldReflection.optimize(Accessor.java:255)
        at com.sun.xml.bind.v2.runtime.property.SingleElementNodeProperty.<init>(SingleElementNodeProperty.java:64)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:77)
        at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
        at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:480)
        at com.sun.xml.bind.v2.runtime.property.PropertyFactory.create(PropertyFactory.java:99)
        at com.sun.xml.bind.v2.runtime.ClassBeanInfoImpl.<init>(ClassBeanInfoImpl.java:150)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.getOrCreate(JAXBContextImpl.java:493)
        at com.sun.xml.bind.v2.runtime.ClassBeanInfoImpl.<init>(ClassBeanInfoImpl.java:137)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.getOrCreate(JAXBContextImpl.java:493)
        at com.sun.xml.bind.v2.runtime.ClassBeanInfoImpl.<init>(ClassBeanInfoImpl.java:137)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.getOrCreate(JAXBContextImpl.java:493)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.<init>(JAXBContextImpl.java:310)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.<init>(JAXBContextImpl.java:109)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl$JAXBContextBuilder.build(JAXBContextImpl.java:1142)
        at com.sun.xml.bind.v2.ContextFactory.createContext(ContextFactory.java:141)
        at com.sun.xml.bind.v2.JAXBContextFactory.createContext(JAXBContextFactory.java:35)
        at javax.xml.bind.ContextFinder.find(ContextFinder.java:358)
        at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:691)
        at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:632)
        at org.eclipse.milo.opcua.binaryschema.generator.DataTypeDictionaryGenerator.writeToOutputStream(DataTypeDictionaryGenerator.java:108)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.writeDictionaryToStream(DataTypeDictionaryManager.java:403)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.writeDictionaryToFile(DataTypeDictionaryManager.java:365)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$null$0(DataTypeDictionaryManager.java:129)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.maybeCompute(Lazy.java:50)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.getOrCompute(Lazy.java:36)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$startup$1(DataTypeDictionaryManager.java:127)
        at org.eclipse.milo.opcua.sdk.server.nodes.filters.AttributeFilters$1.getAttribute(AttributeFilters.java:30)
        at org.eclipse.milo.opcua.sdk.server.nodes.filters.AttributeFilterChain.getAttribute(AttributeFilterChain.java:82)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaNode.getAttribute(UaNode.java:647)
        at org.eclipse.milo.opcua.sdk.server.util.AttributeReader.readAttribute(AttributeReader.java:96)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.readAttribute(UaServerNode.java:114)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.lambda$readAttribute$0(UaServerNode.java:91)
        at java.base/java.util.Optional.map(Optional.java:260)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.readAttribute(UaServerNode.java:91)
        at org.eclipse.milo.opcua.sdk.server.api.ManagedAddressSpace.read(ManagedAddressSpace.java:149)
        at de.cloos.cgate.opcua.adapter.OpcUaObserverNamespace.read(OpcUaObserverNamespace.java:322)
        at org.eclipse.milo.opcua.sdk.server.api.AddressSpaceComposite.lambda$null$16(AddressSpaceComposite.java:330)
        at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.lambda$groupMapCollate$3(GroupMapCollate.java:47)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
        at java.base/java.util.HashMap$EntrySpliterator.forEachRemaining(HashMap.java:1850)
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:575)
        at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
        at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:616)
        at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.allOf(GroupMapCollate.java:77)
        at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.groupMapCollate(GroupMapCollate.java:68)
        at org.eclipse.milo.opcua.sdk.server.api.AddressSpaceComposite.read(AddressSpaceComposite.java:317)
        at org.eclipse.milo.opcua.sdk.server.services.DefaultAttributeServiceSet.onRead(DefaultAttributeServiceSet.java:71)
        at org.eclipse.milo.opcua.sdk.server.SessionManager.onRead(SessionManager.java:871)
        at org.eclipse.milo.opcua.stack.server.UaStackServer.handleServiceRequest(UaStackServer.java:313)
        at org.eclipse.milo.opcua.stack.server.UaStackServer.lambda$onServiceRequest$10(UaStackServer.java:276)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
[... 60 lines cut ...]
        at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
        at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:480)
        at com.sun.xml.bind.v2.runtime.property.PropertyFactory.create(PropertyFactory.java:99)
        at com.sun.xml.bind.v2.runtime.ClassBeanInfoImpl.<init>(ClassBeanInfoImpl.java:150)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.getOrCreate(JAXBContextImpl.java:493)
        at com.sun.xml.bind.v2.runtime.ClassBeanInfoImpl.<init>(ClassBeanInfoImpl.java:137)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.getOrCreate(JAXBContextImpl.java:493)
        at com.sun.xml.bind.v2.runtime.ClassBeanInfoImpl.<init>(ClassBeanInfoImpl.java:137)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.getOrCreate(JAXBContextImpl.java:493)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.<init>(JAXBContextImpl.java:310)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl.<init>(JAXBContextImpl.java:109)
        at com.sun.xml.bind.v2.runtime.JAXBContextImpl$JAXBContextBuilder.build(JAXBContextImpl.java:1142)
        at com.sun.xml.bind.v2.ContextFactory.createContext(ContextFactory.java:141)
        at com.sun.xml.bind.v2.JAXBContextFactory.createContext(JAXBContextFactory.java:35)
        at javax.xml.bind.ContextFinder.find(ContextFinder.java:358)
        at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:691)
        at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:632)
        at org.eclipse.milo.opcua.binaryschema.generator.DataTypeDictionaryGenerator.writeToOutputStream(DataTypeDictionaryGenerator.java:108)
        ... 35 more
```
### Solution
I managed to solve this, by forcing gradle to use version 2.3.6 of org.glassfish.jaxb:jaxb-runtime. Could you please upgrade the dependencies as well?

### Context
Java 17.0.2 (OpenJDK), Milo 0.6.4

kevinherron, 2022-04-14:
@cloos-stone can you verify #963 fixes the issue for you?

kevinherron, 2022-04-14:
Not that these upgrades are dangerous, but I'm somewhat concerned that I'm unable to reproduce this using the ExampleServer on JDK 17, which also has a custom type and writes a dictionary file using `DataTypeDictionaryGenerator`.

closed 2022-04-15 by kevinherron

cloos-stone, 2022-04-25:
Hello @kevinherron ,
sorry for my late reply. I was on vacation for one week and didn't check my mails. 
Thanks for upgrading the dependencies and pushing the changes to master. I checked the new tag that includes the changes and everything works fine. Thanks!

PR #963 (2022-04-15): Update jaxb-runtime and jakarta.activation dependencies
This should fix an error using DataTypeDictionaryGenerator on JDK 17.

fixes #962
commit 150c16bb (2022-04-15): Update jaxb-runtime and jakarta.activation dependencies (#963)

fixes #962