eclipse-milo/milo #964: NPE when adding custom enum field to custom struct

alw-iwu, 2022-04-19:
**Describe the bug**
When generating a datatype dictionary with a custom struct, the DataTypeDictionaryGenerator checks that each field is either in ns=0 or has a Reference to a Default-Binary-Encoding node. This is realized via a [function handed from the DataTypeDictionaryManager](https://github.com/eclipse/milo/blob/7c54cfa30a3e31012e70008d446a17699f0c8775/opc-ua-sdk/dictionary-manager/src/main/java/org/eclipse/milo/opcua/sdk/server/dtd/DataTypeDictionaryManager.java#L449) refering to the HAS_ENCODING_PREDICATE. Custom enums by definition have no encoding nodes. That's why checking for it is unnecessary.

**Expected behavior**
Add struct with enum field. 

**Logs and Packet Captures**
```
11:53:52.849 [milo-shared-thread-pool-3] WARN  o.e.m.o.s.s.d.DataTypeDictionaryManager - Failed to add StructureDescription: QualifiedName{name=ServiceResultType, namespaceIndex=2}
java.lang.NullPointerException: dataTypeEncodingNode for dataTypeId=NodeId{ns=2, id=3004}
	at com.google.common.base.Preconditions.checkNotNull(Preconditions.java:907)
	at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$newDictionaryGenerator$8(DataTypeDictionaryManager.java:455)
	at org.eclipse.milo.opcua.binaryschema.generator.DataTypeDictionaryGenerator.createStructuredType(DataTypeDictionaryGenerator.java:228)
	at org.eclipse.milo.opcua.binaryschema.generator.DataTypeDictionaryGenerator.addStructureDescription(DataTypeDictionaryGenerator.java:88)
	at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$writeDictionaryToStream$3(DataTypeDictionaryManager.java:397)
	at java.base/java.util.concurrent.ConcurrentHashMap$ValuesView.forEach(ConcurrentHashMap.java:4772)
	at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.writeDictionaryToStream(DataTypeDictionaryManager.java:395)
	at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.writeDictionaryToFile(DataTypeDictionaryManager.java:365)
	at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$null$0(DataTypeDictionaryManager.java:129)
	at org.eclipse.milo.opcua.stack.core.util.Lazy.maybeCompute(Lazy.java:50)
	at org.eclipse.milo.opcua.stack.core.util.Lazy.getOrCompute(Lazy.java:36)
	at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$startup$1(DataTypeDictionaryManager.java:127)
	at org.eclipse.milo.opcua.sdk.server.nodes.filters.AttributeFilters$1.getAttribute(AttributeFilters.java:30)
	at org.eclipse.milo.opcua.sdk.server.nodes.filters.AttributeFilterChain.getAttribute(AttributeFilterChain.java:82)
	at org.eclipse.milo.opcua.sdk.server.nodes.UaNode.getAttribute(UaNode.java:647)
	at org.eclipse.milo.opcua.sdk.server.util.AttributeReader.readAttribute(AttributeReader.java:96)
	at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.readAttribute(UaServerNode.java:114)
	at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.lambda$readAttribute$0(UaServerNode.java:91)
	at java.base/java.util.Optional.map(Optional.java:265)
	at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.readAttribute(UaServerNode.java:91)
	at org.eclipse.milo.opcua.sdk.server.api.ManagedAddressSpace.read(ManagedAddressSpace.java:149)
	at org.eclipse.milo.opcua.sdk.server.api.AddressSpaceComposite.lambda$null$16(AddressSpaceComposite.java:330)
	at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.lambda$groupMapCollate$3(GroupMapCollate.java:47)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:195)
	at java.base/java.util.HashMap$EntrySpliterator.forEachRemaining(HashMap.java:1764)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:550)
	at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(AbstractPipeline.java:260)
	at java.base/java.util.stream.ReferencePipeline.toArray(ReferencePipeline.java:517)
	at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.allOf(GroupMapCollate.java:77)
	at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.groupMapCollate(GroupMapCollate.java:68)
	at org.eclipse.milo.opcua.sdk.server.api.AddressSpaceComposite.read(AddressSpaceComposite.java:317)
	at org.eclipse.milo.opcua.sdk.server.services.DefaultAttributeServiceSet.onRead(DefaultAttributeServiceSet.java:71)
	at org.eclipse.milo.opcua.sdk.server.SessionManager.onRead(SessionManager.java:871)
	at org.eclipse.milo.opcua.stack.server.UaStackServer.handleServiceRequest(UaStackServer.java:313)
	at org.eclipse.milo.opcua.stack.server.UaStackServer.lambda$onServiceRequest$10(UaStackServer.java:276)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)

```

**Additional context**
Milo 0.6.4, Java 11

kevinherron, 2022-04-20:
@alw-iwu can you test this branch when you have a chance?

alw-iwu, 2022-04-21:
Works like a charm! Thanks for the quick fix. I'm excited to see how the DataType handling changes with the implementation of 1.04.

kevinherron, 2022-04-21:
> I'm excited to see how the DataType handling changes with the implementation of 1.04.

I imagine it's going to be a lot easier. You'll still use these structures that I "borrowed" from 1.04 to define a struct, but instead of this insane XML dictionary that needs to be generated you just set a `DataTypeDefinition` struct on the new "DataTypeDefinition" attribute that DataType Nodes have starting in 1.04.

commit 0f159401 (2022-04-20): Fix NPE generating struct with enum field

When generating the BSD XML for a struct that has an enum type field
assume the namespace URI is that of the current dictionary. Structs
are not currently defined with enough extra information to look up the
namespace URI; consequently a struct definition may not reference an
enum type defined in another dictionary.

fixes #964
PR #967 (2022-04-25): Fix NPE generating struct with enum field
When generating the BSD XML for a struct that has an enum type field
assume the namespace URI is that of the current dictionary. Enumerations
are not currently defined with enough extra information to look up the
namespace URI; consequently a struct definition may not reference an
enum type defined in another dictionary.

fixes #964
commit 989c9a32 (2022-04-25): Fix NPE generating struct with enum field (#967)

When generating the BSD XML for a struct that has an enum type field
assume the namespace URI is that of the current dictionary. Enums
are not currently defined with enough extra information to look up the
namespace URI; consequently a struct definition may not reference an
enum type defined in another dictionary.

fixes #964
closed 2022-04-25 by kevinherron