eclipse-milo/milo #975: client maxByteStringLength

lukeSky3434, 2022-04-28:
Hello, 

I need to set the **maxByteStringLength** for the client. Is that possible?

Maybe I have to pass the _EndpointConfiguration_ when creating the client?

```
EndpointConfiguration conf = EndpointConfiguration.builder()
  .maxByteStringLength(1024 * 1024 * 4)
  .build();
```

Do you have any suggestions?

lukeSky3434, 2022-04-28:
I also checked out the prosys opcua client library and there you can set the MaxByteStringLength. You can check out the article there you can find the setting:

https://www.prosysopc.com/blog/subscribing-to-a-very-large-number-of-nodes-in-a-prosys-opc-ua-java-sdk-based-server/

kevinherron, 2022-04-28:
Sorry, those settings don't exist for the client in this library. The spec only defines them for the server, and I'm not sure how useful they would be in the client.

lukeSky3434, 2022-04-28:
Ok, I had the same issue like the guy there: https://forum.prosysopc.com/forum/opc-ua-java-sdk/serviceexception-bad_encodinglimitsexceeded/

So I am going to try the EncodingLimits as well, maybe it will solve my issue

lukeSky3434, 2022-04-28:
I am trying to load a xml file from a Node:

```
NodeId translations = new NodeId(1, "SYSTEM.TRANSLATIONS.en");
DataValue value = client.readValue(Double.MIN_VALUE, null, translations).get();
```

kevinherron, 2022-04-28:
Okay... I don't see how this is relevant though.

lukeSky3434, 2022-04-29:
The exact exception:

```
3:17.036 [milo-shared-thread-pool-0] WARN  io.netty.util.ReferenceCountUtil - Failed to release a message: UnpooledSlicedByteBuf(freed)
io.netty.util.IllegalReferenceCountException: refCnt: 0, decrement: 1
	at io.netty.util.internal.ReferenceCountUpdater.toLiveRealRefCnt(ReferenceCountUpdater.java:74) ~[netty-common-4.1.75.Final.jar:4.1.75.Final]
	at io.netty.util.internal.ReferenceCountUpdater.release(ReferenceCountUpdater.java:138) ~[netty-common-4.1.75.Final.jar:4.1.75.Final]
	at io.netty.buffer.AbstractReferenceCountedByteBuf.release(AbstractReferenceCountedByteBuf.java:100) ~[netty-buffer-4.1.75.Final.jar:4.1.75.Final]
	at io.netty.buffer.AbstractDerivedByteBuf.release0(AbstractDerivedByteBuf.java:98) ~[netty-buffer-4.1.75.Final.jar:4.1.75.Final]
	at io.netty.buffer.AbstractDerivedByteBuf.release(AbstractDerivedByteBuf.java:94) ~[netty-buffer-4.1.75.Final.jar:4.1.75.Final]
	at io.netty.util.ReferenceCountUtil.release(ReferenceCountUtil.java:90) ~[netty-common-4.1.75.Final.jar:4.1.75.Final]
	at io.netty.util.ReferenceCountUtil.safeRelease(ReferenceCountUtil.java:116) ~[netty-common-4.1.75.Final.jar:4.1.75.Final]
	at java.util.ArrayList.forEach(ArrayList.java:1259) ~[?:1.8.0_271]
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:96) ~[stack-core-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:77) ~[stack-core-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler.lambda$onSecureMessage$14(UascClientMessageHandler.java:623) ~[stack-client-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:56) ~[stack-core-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119) ~[stack-core-0.6.5.jar:0.6.5]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) ~[?:1.8.0_271]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) ~[?:1.8.0_271]
	at java.lang.Thread.run(Thread.java:748) ~[?:1.8.0_271]
```

lukeSky3434, 2022-05-01:
@kevinherron :

You are right, here it is:

```
08:17:12.977 [milo-shared-thread-pool-2] ERROR org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler - Error decoding symmetric message
org.eclipse.milo.opcua.stack.core.channel.MessageDecodeException: UaException: status=Bad_SecurityChecksFailed, message=expected sequence number 39 but received 38
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:97) ~[stack-core-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decodeSymmetric(ChunkDecoder.java:77) ~[stack-core-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler.lambda$onSecureMessage$14(UascClientMessageHandler.java:623) ~[stack-client-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:56) ~[stack-core-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$Task.run(ExecutionQueue.java:119) ~[stack-core-0.6.5.jar:0.6.5]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) ~[?:1.8.0_271]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) ~[?:1.8.0_271]
	at java.lang.Thread.run(Thread.java:748) ~[?:1.8.0_271]
Caused by: org.eclipse.milo.opcua.stack.core.UaException: expected sequence number 39 but received 38
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder$AbstractDecoder.decode(ChunkDecoder.java:203) ~[stack-core-0.6.5.jar:0.6.5]
	at org.eclipse.milo.opcua.stack.core.channel.ChunkDecoder.decode(ChunkDecoder.java:89) ~[stack-core-0.6.5.jar:0.6.5]
	... 7 more
```

wiresharek cutout:
[image]

lukeSky3434, 2022-05-01:
I played around with the encoding limits again and figured out when I double the 'DEFAULT_MAX_MESSAGE_SIZE' I am getting the data value of the node.

```
new EncodingLimits(
  EncodingLimits.DEFAULT_MAX_CHUNK_SIZE,
  EncodingLimits.DEFAULT_MAX_CHUNK_COUNT,
  EncodingLimits.DEFAULT_MAX_MESSAGE_SIZE * 2,
  EncodingLimits.DEFAULT_MAX_RECURSION_DEPTH))

```

kevinherron, 2022-05-01:
> UaException: expected sequence number 39 but received 38

This implies you are triggering a bug in the server... does it happen every time? What server is it? Can you email me the Wireshark capture or upload it somewhere I can access it?

lukeSky3434, 2022-05-03:
You should have an email in your inbox

kevinherron, 2022-05-03:
I don't see anything, even in spam folder. Maybe gmail has blocked it because of the attachment?