eclipse-milo/milo #995: Adding new ReferenceType does not always work

achintya-kumar, 2022-05-13:
Hi Kevin! 

I am working on adding some ReferenceTypeNode(s) from v1.04 for some experiments. 
I am trying to add the following nodes to the latest version of Milo from master branch:
- HasAddIn
- HasInterface

I started with *HasAddin* and made the following changes:
[image]

[image]

[image]

With these changes, I rebuilt the project.

---
In order to test, we have a [sample OPC UA server](https://github.com/umati/Sample-Server) implementing v1.04 located here: `opc.tcp://opcua.umati.app:4840`
The node of interest w.r.t. *HasAddIn* is called **BasicGMS** with NodeId `ns=18;i=66382`.

[image]

---
Now I try to check if the references can be loaded upon browsing the node as follows:
[image]

The behavior of the client is exactly as expected, so far.

---

Now comes the ***problematic*** part.
I made the changes to incorporate *HasInterface* ReferenceTypeNode into Milo. The changes made are as follows:
[image]

[image]

[image]

A project rebuild is thereafter carried out.

---

The node of interest w.r.t. *HasInterface* is called **Identification** with NodeId `ns=18;i=58148`.
[image]

---

Now I try again to check if the references can be loaded upon browsing the node as follows:
[image]

The *HasInterface*-type references could not be loaded upon trying to browse NodeId `ns=18;i=58148` despite the fact that the node has 4 of such references. 

I am confused as to why this is happening. Could you please assist me with this?

FYI:

- Milo in use: [link](https://github.com/eclipse/milo/commit/11864af3c6cc85ff5b2daa94e1ac5a03d84aa215)
- OPC UA client: Prosys OPC UA Browser v4.1.2-364

kevinherron, 2022-05-13:
Hi,

From a quick look at what you’ve sent, it seems the NodeClassMask you are using is only for Variable and Object Nodes, which would exclude the targets of any HasInterface reference.

If that’s not the issue I’ll have to take a closer look on Monday. I’ll be away for the next 3 days.

achintya-kumar, 2022-05-13:
Thanks as always, Kevin! 
The problem is now resolved. I had a feeling this is caused by the NodeClassMask :)

[image]

closed 2022-05-13 by achintya-kumar