eclipse-milo/milo #1002: DatatypeConverter Class deprecation

hutcheb, 2022-05-23:
After finding an issue compiling the server stack using openjdk 17, milo 0.6.6

Exception in thread "main" java.lang.NoClassDefFoundError: javax/xml/bind/DatatypeConverter
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaXmlStreamDecoder.readByteString(OpcUaXmlStreamDecoder.java:332)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaXmlStreamDecoder.readBuiltinType(OpcUaXmlStreamDecoder.java:723)

It looks like the DatatypeConverter class seems to have been deprecated.
https://javadoc.io/doc/org.glassfish.jaxb/jaxb-runtime/2.3.6/org.glassfish.jaxb.runtime/com/sun/xml/bind/package-summary.html

closed 2022-05-23 by hutcheb

kevinherron, 2022-05-24:
@hutcheb what were you doing when you encountered this?

hutcheb, 2022-05-24:
Hi Kevin,  ignore most of the issue I raised, I don’t really know what the issue is yet.

I was running the PLC4X - Milo integration server, on startup this error comes up. Although I was able to confirm that the jaxb api is included in the milo stack thru the glass fish-runtime-> Jakarta api dependencies so not to sure.

I should have some time tonight to look into it further.

kevinherron, 2022-05-24:
Okay let me know what you find. You're right that `javax.xml.bind.DatatypeConverter` should be on the class path already as part of `jakarta.xml.bind-api`:
```
[INFO] +- org.glassfish.jaxb:jaxb-runtime:jar:2.3.6:compile
[INFO] |  +- jakarta.xml.bind:jakarta.xml.bind-api:jar:2.3.3:compile
```

Milo is compiled with JDK 8, but should run on JDK 11 or JDK 17 without issue.

hutcheb, 2022-05-24:
Something to do with

<dependency>
      <groupId>javax.xml.bind</groupId>
      <artifactId>jaxb-api</artifactId>
      <version>2.3.1</version>
    </dependency>

Including this fixes it.

hutcheb, 2022-05-24:
Might be worth updating references to use jakata.xml.bind instead of javax.xml.bind?

kevinherron, 2022-05-24:
Hmm. I'm not sure why you need that. Running the demo server on JDK 17 doesn't produce any kind of error like that, though perhaps your client?/server? is going down some code path the demo server is not.

hutcheb, 2022-05-24:
I found the issue, we had the jaxb-runtime pinned at 3.0.2 for some reason.

Thanks for listening to me blabber ;)

kevinherron, 2022-05-24:
I thought it might be something like that when you mentioned `jakata.xml.bind`. Glad you got it working.