eclipse-milo/milo #1023: DataTypeDictionaryManager not found

ronytesler, 2022-07-26:
I added <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>sdk-server</artifactId>
            <version>0.6.7</version>
        </dependency>
like in the instructions.
I try to run the server example, where the dependency is '0.6.8-SNAPSHOT', and org.eclipse.milo.opcua.sdk.server.dtd is not found.

kevinherron, 2022-07-26:
Have you run `mvn clean install` at the project root?

ronytesler, 2022-07-26:
Yes

kevinherron, 2022-07-26:
I need more information about what you're trying to do. Do you have the Milo project open an in IDE and trying to run the example, or something else?

kevinherron, 2022-07-26:
It sounds like you may just be missing the `dictionary-manager` dependency: 

https://github.com/eclipse/milo/blob/1e063b9fc880e4b10ee6fcbd2b665d2e7eaed254/milo-examples/server-examples/pom.xml#L18-L28

ronytesler, 2022-07-26:
yes, exactly.
Cannot resolve org.eclipse.milo:dictionary-manager:0.6.8-SNAPSHOT
If I try to use 0.6.7, I get this:
Cannot resolve org.eclipse.milo:stack-core:0.6.7

ronytesler, 2022-07-26:
And this:
Could not transfer artifact org.eclipse.milo:dictionary-manager:pom:0.6.7 from/to central (https://repo.maven.apache.org/maven2): Transfer failed for https://repo.maven.apache.org/maven2/org/eclipse/milo/dictionary-manager/0.6.7/dictionary-manager-0.6.7.pom

ronytesler, 2022-07-26:
The first one I get when I do 'reload project', and the second one is for 'install'.

kevinherron, 2022-07-26:
Okay, well this sounds like a problem with your network or environment, not something I can fix for you. See if you can run Maven or the IDE task with more debug information. Why is the transfer failing?

The artifact is in Maven Central: https://repo1.maven.org/maven2/org/eclipse/milo/dictionary-manager/0.6.7/

ronytesler, 2022-07-26:
It was a network problem, thanks!

closed 2022-07-26 by ronytesler