eclipse-milo/milo #1035: Caused by: javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.

jnsw, 2022-08-31:
I am using your library and creating a Jar-file with all dependecies (with Proguard). If I am running it and want to connect it, I get following error. The other issues doesn't resolve it. Can I somewhere explicit say which implementation of Jaxb my jar needs to use?

```
[milo-shared-thread-pool-7] WARN org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager - Failed to write dictionary file
java.lang.RuntimeException: failed to write dictionary file
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$null$0(DataTypeDictionaryManager.java:131)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.maybeCompute(Lazy.java:50)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.getOrCompute(Lazy.java:36)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$startup$1(DataTypeDictionaryManager.java:127)
        at org.eclipse.milo.opcua.sdk.server.nodes.filters.AttributeFilters$1.getAttribute(AttributeFilters.java:30)
        at org.eclipse.milo.opcua.sdk.server.nodes.filters.AttributeFilterChain.getAttribute(AttributeFilterChain.java:82)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaNode.getAttribute(UaNode.java:647)
        at org.eclipse.milo.opcua.sdk.server.util.AttributeReader.readAttribute(AttributeReader.java:96)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.readAttribute(UaServerNode.java:114)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.lambda$readAttribute$0(UaServerNode.java:91)
        at java.base/java.util.Optional.map(Unknown Source)
        at org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode.readAttribute(UaServerNode.java:91)
        at org.eclipse.milo.opcua.sdk.server.api.ManagedAddressSpace.read(ManagedAddressSpace.java:148)
        at org.eclipse.milo.opcua.sdk.server.api.AddressSpaceComposite.lambda$null$16(AddressSpaceComposite.java:330)
        at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.lambda$groupMapCollate$3(GroupMapCollate.java:47)
        at java.base/java.util.stream.ReferencePipeline$3$1.accept(Unknown Source)
        at java.base/java.util.HashMap$EntrySpliterator.forEachRemaining(Unknown Source)
        at java.base/java.util.stream.AbstractPipeline.copyInto(Unknown Source)
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
        at java.base/java.util.stream.AbstractPipeline.evaluate(Unknown Source)
        at java.base/java.util.stream.AbstractPipeline.evaluateToArrayNode(Unknown Source)
        at java.base/java.util.stream.ReferencePipeline.toArray(Unknown Source)
        at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.allOf(GroupMapCollate.java:77)
        at org.eclipse.milo.opcua.sdk.core.util.GroupMapCollate.groupMapCollate(GroupMapCollate.java:68)
        at org.eclipse.milo.opcua.sdk.server.api.AddressSpaceComposite.read(AddressSpaceComposite.java:317)
        at org.eclipse.milo.opcua.sdk.server.services.DefaultAttributeServiceSet.onRead(DefaultAttributeServiceSet.java:71)
        at org.eclipse.milo.opcua.sdk.server.SessionManager.onRead(SessionManager.java:871)
        at org.eclipse.milo.opcua.stack.server.UaStackServer.handleServiceRequest(UaStackServer.java:313)
        at org.eclipse.milo.opcua.stack.server.UaStackServer.lambda$onServiceRequest$10(UaStackServer.java:276)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)
Caused by: java.io.IOException: failed to write dictionary to OutputStream
        at org.eclipse.milo.opcua.binaryschema.generator.DataTypeDictionaryGenerator.writeToOutputStream(DataTypeDictionaryGenerator.java:123)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.writeDictionaryToStream(DataTypeDictionaryManager.java:403)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.writeDictionaryToFile(DataTypeDictionaryManager.java:365)
        at org.eclipse.milo.opcua.sdk.server.dtd.DataTypeDictionaryManager.lambda$null$0(DataTypeDictionaryManager.java:129)
        ... 31 more
Caused by: javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
 - with linked exception:
[java.lang.ClassNotFoundException: com.sun.xml.internal.bind.v2.ContextFactory]
        at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:232)
        at javax.xml.bind.ContextFinder.find(ContextFinder.java:375)
        at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:691)
        at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:632)
        at org.eclipse.milo.opcua.binaryschema.generator.DataTypeDictionaryGenerator.writeToOutputStream(DataTypeDictionaryGenerator.java:108)
        ... 34 more
Caused by: java.lang.ClassNotFoundException: com.sun.xml.internal.bind.v2.ContextFactory
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(Unknown Source)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(Unknown Source)
        at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
        at javax.xml.bind.ServiceLoaderUtil.nullSafeLoadClass(ServiceLoaderUtil.java:92)
        at javax.xml.bind.ServiceLoaderUtil.safeLoadClass(ServiceLoaderUtil.java:125)
        at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:230)
        ... 38 more
```

kevinherron, 2022-08-31:
I think these are the relevant dependencies: https://github.com/eclipse/milo/blob/938ed51db759872a87d98543561027b4766f2c9e/opc-ua-stack/bsd-core/pom.xml#L24-L34

I've never used Proguard, I don't know how to tell it to exclude these from obfuscation or minimization, but I suspect it's because there is some reflection-based access happening that causes Proguard to prune classes it thinks are unused.

jnsw, 2022-09-01:
Okay, this seems a too specific problem, because I already did that and all other solution I could found online, even with a hacky solution implementing the missing class with the already present implementation of `com.sun.xml.bind.v2.ContextFactory` by extending it. If I have a solution, I will come back.