eclipse-milo/milo #1058: Can I need manually reconnect when  seviceFault was happen?

zhengshubin, 2022-09-27:
**Describe the bug**
A clear and concise description of behavior being reported and, if applicable, why it's incorrect.
I had add faultLisener  and when it  occur  serviceFault, I had  manually  disconnect  and reconnect it. Is it right?
Can it reconnect  automatically？ Here is my code .Thanks.

client.addFaultListener(
	(serviceFault)->{
		logger.error(serviceFault.toString());
		runningServerMap.remove(machine);
		client.disconnect();
		reconnectServerSet.add(machine);
		
});
**Expected behavior**
A clear and concise description of what you expected to happen instead.

**Logs and Packet Captures**
If applicable, add application logs or Wireshark captures to help explain your problem.

**Additional context**
Version of Eclipse Milo you are using, OPC UA client or software server you are interacting with, etc...

kevinherron, 2022-09-27:
You do not need to manually reconnect, the library handles it automatically.

All you need to do is add a SubscriptionListener to the OpcUaSubscriptionManager and implement SubscriptionListener::onSubscriptionTransferFailed. Any time this is called it means the client has reconnected but was unable to resume the session or transfer the old subscription, so you have to re-create it.

closed 2022-09-27 by kevinherron

cityWalker312, 2022-11-08:
> You do not need to manually reconnect, the library handles it automatically.
>
> All you need to do is add a SubscriptionListener to the OpcUaSubscriptionManager and implement [...]

"re-create it" means we should disconnect client , create a new client and connect the server use new client.
Am I correct in my understanding?

kevinherron, 2022-11-08:
@cityWalker312 No, re-create the subscription. You don't need to do anything with the client.

cityWalker312, 2022-11-08:
> @cityWalker312 No, re-create the subscription. You don't need to do anything with the client.

Thanks :)