eclipse-milo/milo #1062: Continuous return Bad_Timeout in SubscriptionListener.onPublishFailure

fonlin, 2022-10-12:
The SubscriptionListener.onPublishFailure continuous report exception `Bad_Timeout`. However, `Subscription` looks normal and client can receive changed data at the same time.

I guess it has something to do with `PublishRequest Timeout Hint`. 

[image]

[image]

fonlin, 2022-10-12:
I found a issue #679 , but i has checked Kepware and my client timestamp, and they are similar

kevinherron, 2022-10-12:
Can you get a Wireshark capture?

This is possibly an issue with the server you're connected to and how it's handling receiving multiple PublishRequests. Some are timing out, but some are not, which is why your subscription continues to function.

fonlin, 2022-10-13:
> Can you get a Wireshark capture?
> 
> This is possibly an issue with the server you're connected to and how it's handling receiving multiple PublishRequests. Some are timing out, but some are not, which is why your subscription continues to function.
> 
> I would imagine this is reproducible given the number of subscriptions you have created and the timing properties of each, so those details would be relevant too. So would the vendor and version of the server you are connected to.

OK, We have upgraded the Kepware version and will continue to observe this issue. Thanks a lot for your answer!