eclipse-milo/milo #1069: Can ua client accept the server certificate automatically?

zyeewang, 2022-10-26:
For now, I must copy a .der file to the client\security\pki\trusted\certs manually. Does it doable to make the client accept the server certificate automatically?

kevinherron, 2022-10-26:
You can configure the client's `CertificateValidator` as an instance of `ClientCertificateValidator.InsecureValidator`:

https://github.com/eclipse/milo/blob/3419235853491ac89f3391bffd337c8200522522/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/security/ClientCertificateValidator.java#L29-L52

It's not recommended that you do this for anything but testing. A production application should always require certificates are trusted.

kevinherron, 2022-10-26:
Alternatively, when you get the server `EndpointDescription`s, the certificate is available in that structure, and you can trust it by calling `DefaultCertificateManager::addTrustedCertificate`.

zyeewang, 2022-10-27:
> 

May I know how to utilize DefaultCertificateManager in client? Only server example involve some code about DefaultCertificateManager.

zyeewang, 2022-10-27:
> Alternatively, when you get the server `EndpointDescription`s, the certificate is available in that structure, and you can trust it by calling `DefaultCertificateManager::addTrustedCertificate`.

Actually, I am a little confused about how to utilize EndPointDescription to get the server's certificate. Would you mind offer a demo?