eclipse-milo/milo #1073: Call milo opc-ua server ConditionType_ConditionRefresh method failed.

cityWalker312, 2022-10-27:
**Describe the bug**
Use milo client call milo opc-ua server ConditionType_ConditionRefresh failed.
We try to call the method from UaExpert app failed too on milo opc-ua server.

Use milo client call open62541 opc-ua server ConditionType_ConditionRefresh successfully.

Here is the call method codeblock.
```java
private boolean callConditionRefresh(OpcUaClient client, CompletableFuture<OpcUaClient> future, int subId) {
        NodeId objectId = Identifiers.ConditionType;
        NodeId methodId = Identifiers.ConditionType_ConditionRefresh;
        CallMethodRequest request = new CallMethodRequest(
                objectId,
                methodId,
                new Variant[]{new Variant(UInteger.valueOf(subId))}
        );
        client.call(request).thenCompose(result -> {
            StatusCode statusCode = result.getStatusCode();
            if (statusCode.isGood()) {
                return CompletableFuture.completedFuture(0);
            } else {
                StatusCode[] inputArgumentResults = result.getInputArgumentResults();
                if (inputArgumentResults != null) {
                    for (int i = 0; i < inputArgumentResults.length; i++) {
                        logger.error("inputArgumentResults[{}]={}", i, inputArgumentResults[i]);
                    }
                }
                CompletableFuture<Integer> f = new CompletableFuture<>();
                f.completeExceptionally(new UaException(statusCode));
                return f;
            }
        });

        return true;
    }
```

**Expected behavior**
A clear and concise description of what you expected to happen instead.

**Logs and Packet Captures**
If applicable, add application logs or Wireshark captures to help explain your problem.
[image]

**Additional context**
Version of Eclipse Milo you are using, OPC UA client or software server you are interacting with, etc...

kevinherron, 2022-10-28:
Thanks for the report, this does look like a bug in the Milo server. 

Note to future self: the issue is "known" and will be fixed when this TODO is resolved:
https://github.com/eclipse/milo/blob/3419235853491ac89f3391bffd337c8200522522/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/api/methods/AbstractMethodInvocationHandler.java#L79-L83

cityWalker312, 2022-10-29:
> Thanks for the report, this does look like a bug in the Milo server.
>
> Note to future self: the issue is "known" and will be fixed when this TODO is resolved: [...]

Thanks for your reply. 
Does milo have a plan to fix this issue ? Our  project need to use this feature.

kevinherron, 2024-03-16:
WIP branch with a test that reproduces: https://github.com/eclipse/milo/tree/issues/gh1073

kevinherron, 2024-03-17:
Fixed in https://github.com/eclipse/milo/commit/73e9fef7b933198471ad25e747c845044d3d5880 for 1.0 release.

PR #1240 (2024-03-17): Input argument datatype validation in AbstractMethodInvocationHandler
Implement new input argument datatype validation in `AbstractMethodInvocationHandler`.

Input arguments of the following types should be supported:
- built-in types
- simple types
- abstract types
- structured types

fixes #1073
commit 73e9fef7 (2024-03-17): Input argument datatype validation in AbstractMethodInvocationHandler (#1240)

Implement new input argument datatype validation in `AbstractMethodInvocationHandler`.

Input arguments of the following types should be supported:
- built-in types
- simple types
- abstract types
- structured types

fixes #1073
closed 2024-03-17 by kevinherron