eclipse-milo/milo #1076: HistoryRead needs more information from the request

ergouser, 2022-11-01:
In DefaultAttributeHistoryServiceSet

HistoryRead is called as:

   server.getAddressSpaceManager().historyRead(
            context,
            details,
            request.getTimestampsToReturn(),
            nodesToRead
        );

More information is required from the request. At a minimum the Client timeout hint is required (https://reference.opcfoundation.org/v104/Core/docs/Part11/6.4.3/#6.4.3.1) to accommodate long history requests.

kevinherron, 2022-11-01:
Hmm, noted. 

Unfortunately, I won't be fixing this in the 0.x release series, but I'm working on 1.0 right now, and on some refactoring of the stack/transport layers in particular, and this seems like an appropriate place to make sure it's fixed. I think I'll try to make sure the `RequestHeader` (which contains the timeout hint) is passed along all the way to the corresponding service calls on `AddressSpace`.

kevinherron, 2022-11-02:
Thinking about it a little more that won't always help. An inbound request can routed to multiple `AddressSpace` implementations based on the `nodesToRead`, and neither of them know how long the other is going to take. The same is true for any kind of inbound request that might want to look at the hint.

ergouser, 2022-11-02:
Can you have more than one result (different results from different address spaces?) If not and multiple address spaces are contributing to the single result they would have to be coordinated in some way to deliver the result and so would also need to coordinate timing to ensure the client doesn't timeout.

kevinherron, 2022-11-02:
Yes, each address space could possibly be contributing results.

You can see how this works (for all the services) in AddressSpaceComposite: https://github.com/eclipse/milo/blob/3419235853491ac89f3391bffd337c8200522522/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/api/AddressSpaceComposite.java#L372-L406

Supporting this will be a tough change. I think it would require something like:
- passing the initial time the request was received along.
- passing the timeout hint along.
- making GroupMapCollate a sequential operation instead of parallel. 

Modifying the behavior of GMC would be a big change with performance consequences, but it would allow each address space as it receives its requested operation slice to look at the hint and how much time has elapsed since the request was received and determine if it has enough time, or any time at all left, to actually do anything.

Big changes I'll have to think about for 1.0...

ergouser, 2022-11-02:
I suspect that there are many details not yet considered in the case of multiple addressspaces.  The timeout hint is one, continuationpoints also have to be coordinated.  The client will only return one but, presumably, all the addresspaces with more data need to be part of building that.  I suspect that the addresspaces would have to be quite tightly coupled.

kevinherron, 2022-11-02:
I think continuation points may work as is. In the case of history, the target AddressSpace(s) are targeted using the `NodeId` in the `HistoryReadValueId`, and so if the target AddressSpace chose to return a continuation point, the subsequent call with a `HistoryReadValueId` containing a continuation point would be passed to that same AddressSpace.

kevinherron, 2022-11-20:
This information will be available when 1.0 is released via `ServiceOperationContext`, which in the case of history is the `HistoryReadContext` and `HistoryUpdateContext`.

The `auditEntryId`, `timeoutHint`, and `additionalHeader` fields from the `RequestHeader` are now passed through.

closed 2022-11-20 by kevinherron