eclipse-milo/milo #1077: DateTime should allow epochs other than UTC

ergouser, 2022-11-02:
TAI is specifically permitted by the standard:  https://reference.opcfoundation.org/v104/Core/docs/Part6/6.4/  GPS is commonly used for high-precision measurements.

There are applications where having a monotonic clock is essential.

kevinherron, 2022-11-02:
You're free to use whatever fancy time source you want and then convert it to the corresponding Java `Date` or `Instant` and construct a `DateTime` with that.

`DateTime` as encoded on the wire and received by other OPC UA applications is always UTC, there is no way to indicate a different epoch is being used.

ergouser, 2022-11-02:
My reading of https://reference.opcfoundation.org/v104/Core/docs/Part6/6.4/ suggests that the server time would be provided to the client in TAI.  Put another way, the client needs monotonic time and the standard seems to allow this.  There's some ambiguity in that section but:

_Servers that have access to source for International Atomic Time (TAI) may choose to use this instead of UTC._

Seems pretty convincing, but it would be good to get some clarification (anyone we can ask?).  It's mostly useless to have the server in TAI and convert to UTC for the client so, assuming that 6.4 is not intended to be mostly useless (and it's not one of the better sections), encoding TAI is required.  Most of the time you can convert from TAI to UTC, but around the time of leap-seconds everything goes to heck because there are many different opinions on how to handle leap seconds, including rubber-seconds.

Instant.now() without a Clock forces UTC so even if the server is supposed to be maintaining time in TAI that call will break it.

kevinherron, 2022-11-02:
I can ask some other members of the working group, but I think you're reading this wrong and that it _is_ mostly useless. 

It's saying 3 things:
1. OPC UA time is always UTC
2. The source of the server/system time may be something else, like TAI, and that's okay
3. The client shouldn't explode if DateTimes received from the server go backward or forward from expected due to time syncs

Take a look at the encoding for DateTime in Part 6: https://reference.opcfoundation.org/Core/Part6/5.2.2/#5.2.2.5. There is no possibility to indicate to a client that it is anything but 100 nanosecond intervals from UTC epoch.

I'll forward this issue along to someone and see what they have to say.

kevinherron, 2022-11-02:
That said, I could see some value in the ability to set a static/global "Clock" on `DateTime` that it would use when calling `new DateTime()`, or any of the static `now()` variations.

I think this would mostly be useful for testing. I seriously doubt anybody has some Java application-level time source they could plug in and that if they did have a fancy time source it wouldn't be at the OS-level, and just keeping the system clock in accurate, and automatically used when Java gets the time.

ergouser, 2022-11-02:
There's a default clock : 

Clock clock = Clock.systemUTC();

But you're right, if you make it settable it might be handy for testing (as well as TAI).

There's a Java library dedicated to all things time and date - https://github.com/MenoData/Time4J  Timescales are [here](http://www.time4j.net/tutorial/timescales.html)

kevinherron, 2022-11-02:
Here's the response I got back to the email I sent asking about this issue:

> HI Kevin,
>  
> Your response is correct. While servers may use TAI as source for counting time, clients are expected to assume it is UTC and deal with the fact that different servers have different definitions of what the ‘current time’ is.
>  
> See the definition of UtcTime should make it clear that the times are to be interpreted as UTC by the client even if the ultimate sources are different
> https://reference.opcfoundation.org/Core/Part3/8.37/
>  
> “It should be noted that the SourceTimestamp (see OPC 10000-4) and ServerTimestamp (see [OPC 10000-4](https://reference.opcfoundation.org/v105/Core/docs/Part4/)) may originate from different clocks that have no synchronization. It is also possible that one may use UTC while the other uses [TAI](https://reference.opcfoundation.org/v105/Core/docs/Part3/2/#TAI).”
>  
> However, I can see the specification has muddied up the distinction between absolute time and the means to count time from a baseline (e.g. does the counting mechanism add leap seconds or not). In UA we are primarily concerned with counting time from a baseline since applications cannot assume precisely synchronized clocks.

I think the wording in 8.37 even further confuses things, so perhaps there's some clarification we can do to both of these parts in the spec that talk about TAI, but ultimately all time in OPC UA is UTC, and there is no mechanism to use alternate epochs when it comes to the `DateTime` values transferred between client and server.

ergouser, 2022-11-02:
I find that response less elucidating than the spec:

> Your response is correct. 

What was the question?

> While servers may use TAI as source for counting time, clients are expected to assume it is UTC and deal with the fact that different servers have different definitions of what the ‘current time’ is.

Suggests that the server timestamp (on the wire) can be TAI and the client is supposed to just assume that it's UTC even though it isn't.

Somewhat reinforced by this:

> A UA client does not care about the difference between a server using TAI and a server that has a UTC clock that is set 37s ahead of the client’s UTC clock.

Which again says pass TAI as if it were UTC and ignore the 37 seconds.

Finally this:

> “It should be noted that the SourceTimestamp (see [OPC 10000-4](https://reference.opcfoundation.org/v105/Core/docs/Part4/)) and ServerTimestamp (see OPC 10000-4) may originate from different clocks [...]

Which repeats the same mantra, you can pass the Server time as TAI even though the client is sending times in UTC (so there will be a 37 s difference between them if they are perfectly synchronized to their respective clocks).

All of which leads me to the opposite opinion from yours which is that server DateTime should support TAI (or from the UA standpoint any other epoch - "In UA we are primarily concerned with counting time from a baseline" - which really is the strongest argument of all for using TAI - UTC doesn't do that).

ergouser, 2022-11-02:
The clear problem is this:

> All time values conveyed between OPC UA Servers and Clients are UTC values

(https://reference.opcfoundation.org/Core/Part3/8.37/)

Which, if you accept it as true means that TAI has no role in this at all, should not be mentioned in the spec and cannot be use.

If the spec really means that then why even mention TAI.  In reality there is no absolute conversion between TAI and UTC so you can only use one or the other.  That is, there's no point in sync'ing the server to TAI unless you send TAI across the wire.

ergouser, 2022-11-02:
For the record, it's not just OPC that's struggling with this.  The "rubber second" is contentious, and implemented differently by different organization, but no one like minutes that have 61 seconds either.  This leap-smearing is the reason that, around a leap-second, you can't reliably convert TAI to UTC (and that's really the only time you care).  There's a chance that the leap-second will be eliminated, and there's a chance that there will be first negative leap-second...

kevinherron, 2022-11-03:
I'm closing this issue. Feel free to open a Mantis issue to get the spec clarified or propose a change: https://mantis.opcfoundation.org, though you may need to have an OPCF account of some certain level.

It's obvious right now that despite the ambiguities and issues with various passages from the spec that OPC DateTime is always UTC time and there simply isn't a mechanism to convey that a time using any other representation or epoch.

closed 2022-11-03 by kevinherron

kevinherron, 2022-11-03:
FWIW, I don't really understand why there's any explicit mention of TAI either. I think it should stop short at just explaining that clocks may be unsynchronized, between both the client and server, and even between a Server timestamp and Source timestamp within a DataValue, as those may come from entirely different sources.

randy-armstrong, 2022-11-03:
I think the right way to address this is to either allow TAI to specified in the LocalTime Property on DataVariables to to add a new TimeEpoch Property: https://reference.opcfoundation.org/Core/Part3/5.6.2/

This will allow servers to publish that their SourceTimestamps are TAI.

ergouser, 2022-11-23:
FYI. There is a new proposal  (Nov 22) from the General Conference on Weights and Measures (CGPM) to scrap the leap second (https://www.bipm.org/documents/20126/64811223/Resolutions-2022.pdf) starting in 2035 and for at least a century.  Removing leap-seconds from UTC renders this problem mostly moot - without the leap seconds epochs can (largely) be converted.  BUT that's 2035 and there's a thought that there may need to be a negative leap second before that.  There's never been a need for a negative leap-second (there's no general agreement on why the earth's rotation speed has increased) so it's untested an so likely to be a major problem if it's implemented.