eclipse-milo/milo #1083: Unable to read value of Property of Node

ckyara, 2022-11-14:
Hi,
I want to read value of Node property for below record using getProperty(QualifiedProperty<T> property) method of UaNode class. Although, i am able to parse below record and successfully mapped into UaNode object but don't know that how can create QualifiedProperty<T> object from UaNode object to get value of PropertyTypeNoe.

2022-11-10 06:35:14.112 INFO: ParentId: ns=2;s=M|FIKOGATEWAY1, ChildId: ns=2;s=V|FIKOGATEWAY1.bin_disc.SU/FIC-0806TK_LUK.IP_VALUE, **BrowseName: 2:bin_disc.SU/FIC-0806TK_LUK.IP_VALUE**, Value: 1, DataType: ns=0;i=6, NodeClass: Variable, J**avaClass: UaVariableNode**, Description: A fixed area field
2022-11-10 06:37:21.168 INFO: ParentId: ns=2;s=V|FIKOGATEWAY1.bin_disc.SU/FIC-0806TK_LUK.IP_VALUE, ChildId: ns=2;s=A|104|FIKOGATEWAY1.bin_disc.SU/FIC-0806TK_LUK.IP_VALUE, **BrowseName: 2:FIKOGATEWAY1.bin_disc.SU/FIC-0806TK_LUK.IP_VALUE.HiRange**, Value: 1.0, DataType: ns=0;i=11, NodeClass: Variable, **JavaClass: PropertyTypeNode**, Description: High instrument range

kevinherron, 2022-11-14:
That method is meant for use in code-generated code where it has the information to construct a QualifiedProperty ahead of time.

You have the NodeId of the property Node, so just get it from the AddressSpace:
```
UaVariableNode propertyNode = client.getAddressSpace().getVariableNode(ChildId);

DataValue value = propertyNode.getValue(); // or .readValue()
```

ckyara, 2022-11-15:
Thanks Kevin for the needful. It's working now.

closed 2022-11-15 by ckyara