eclipse-milo/milo #1092: CustomStructType decode(SerializationContext context, OpcUaBinaryStreamDecoder decoder) decoder methods not use field name to catch date

EnRiX84, 2022-12-19:
I'm using milo 0.6.8 version.
Follow my CustomStruct codec to decode opcua tree in the screenshot.

```
public static class Codec implements OpcUaBinaryDataTypeCodec<CustomStructType> {

		@Override
		public Class<CustomStructType> getType() {
			return CustomStructType.class;
		}

		@Override
		public CustomStructType decode(SerializationContext context, OpcUaBinaryStreamDecoder decoder) throws UaSerializationException {		
			String Name = decoder.readString();  //OK DECODED --> NomeTest
			System.out.println("Name=" + Name + "|");
			
			Integer DataType = decoder.readByte().intValue();  //OK DECODED --> 2
			System.out.println("DataType=" + DataType + "|");
			
			ByteString StringValue = decoder.readByteString(); //ERROR!!! ERROR c.a.w.protocol.opc.milo.ClientRunner - //Error running client example: readerIndex(17) + length(335544320) exceeds writerIndex(48): UnpooledHeapByteBuf(ridx: 17, widx: //48, cap: 48/48)
			System.out.println("StringValue=" + StringValue + "|");
			
			Integer IntValue = decoder.readByte().intValue();
			System.out.println("IntValue=" + IntValue + "|");
			
			Float FloatValue = decoder.readByte().floatValue();
			System.out.println("Float=" + FloatValue + "|");
						
			return new CustomStructType(Name, 0, "", 0, new Float("1"));
		}

		@Override
		public void encode(SerializationContext context, OpcUaBinaryStreamEncoder encoder, CustomStructType value) throws UaSerializationException {
			encoder.writeString("Name", value.Name);
			encoder.writeString("StringValue", value.StringValue);
			encoder.writeInt32("IntValue", value.IntValue);
			encoder.writeFloat("FloatValue", value.FloatValue);
		}
	}
```

[image]

I don't know why when i use decoder.readByteString(); I receive this error:  ERROR c.a.w.protocol.opc.milo.ClientRunner - Error running client example: readerIndex(17) + length(335544320) exceeds writerIndex(48): UnpooledHeapByteBuf(ridx: 17, widx: 48, cap: 48/48)

What should I use?

Why if i use `decoder.readString("StringValue");` the parameter is ignored?

Thanks to all and thanks for your work and your time

kevinherron, 2022-12-19:
You shouldn't be using `readByteString`, you should be using `readString`. The parameter is ignored because the binary serialization format doesn't actually care about the field names the way XML and JSON do.

EnRiX84, 2022-12-19:
Thank you Kevinherron, very fast!

I have implemented your solution but receive this error: 

```
ERROR c.a.w.protocol.opc.milo.ClientRunner - Error running client example: String index out of range: 335544337
java.lang.StringIndexOutOfBoundsException: String index out of range: 335544337
        at java.lang.String.checkBounds(String.java:385)
        at java.lang.String.<init>(String.java:462)
        at io.netty.buffer.ByteBufUtil.decodeString(ByteBufUtil.java:1276)
        at io.netty.buffer.AbstractByteBuf.toString(AbstractByteBuf.java:1246)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readLengthPrefixedString(OpcUaBinaryStreamDecoder.java:442)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readString(OpcUaBinaryStreamDecoder.java:370)
        at com.alleantia.weblogger.protocol.opc.milo.types.CustomStructType$Codec.decode(CustomStructType.java:138)
        at com.alleantia.weblogger.protocol.opc.milo.types.CustomStructType$Codec.decode(CustomStructType.java:1)
        at org.eclipse.milo.opcua.stack.core.types.OpcUaDefaultBinaryEncoding.decode(OpcUaDefaultBinaryEncoding.java:117)
        at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.lambda$decode$0(ExtensionObject.java:105)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.maybeCompute(Lazy.java:50)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.getOrCompute(Lazy.java:36)
        at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:105)
        at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:96)
        at com.alleantia.weblogger.protocol.opc.milo.ReadWriteCustomDataTypeNodeExample.run(ReadWriteCustomDataTypeNodeExample.java:108)
        at com.alleantia.weblogger.protocol.opc.milo.ClientRunner.run(ClientRunner.java:156)
        at com.alleantia.weblogger.protocol.opc.milo.ReadWriteCustomDataTypeNodeExample.main(ReadWriteCustomDataTypeNodeExample.java:52)
17:15:04.790 [ForkJoinPool.commonPool-worker-1] ERROR c.a.w.protocol.opc.milo.ClientRunner - Error running example: String index out of range: 335544337
java.lang.StringIndexOutOfBoundsException: String index out of range: 335544337
        at java.lang.String.checkBounds(String.java:385)
        at java.lang.String.<init>(String.java:462)
        at io.netty.buffer.ByteBufUtil.decodeString(ByteBufUtil.java:1276)
        at io.netty.buffer.AbstractByteBuf.toString(AbstractByteBuf.java:1246)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readLengthPrefixedString(OpcUaBinaryStreamDecoder.java:442)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readString(OpcUaBinaryStreamDecoder.java:370)
        at com.alleantia.weblogger.protocol.opc.milo.types.CustomStructType$Codec.decode(CustomStructType.java:138)
        at com.alleantia.weblogger.protocol.opc.milo.types.CustomStructType$Codec.decode(CustomStructType.java:1)
        at org.eclipse.milo.opcua.stack.core.types.OpcUaDefaultBinaryEncoding.decode(OpcUaDefaultBinaryEncoding.java:117)
        at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.lambda$decode$0(ExtensionObject.java:105)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.maybeCompute(Lazy.java:50)
        at org.eclipse.milo.opcua.stack.core.util.Lazy.getOrCompute(Lazy.java:36)
        at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:105)
        at org.eclipse.milo.opcua.stack.core.types.builtin.ExtensionObject.decode(ExtensionObject.java:96)
        at com.alleantia.weblogger.protocol.opc.milo.ReadWriteCustomDataTypeNodeExample.run(ReadWriteCustomDataTypeNodeExample.java:108)
        at com.alleantia.weblogger.protocol.opc.milo.ClientRunner.run(ClientRunner.java:156)
        at com.alleantia.weblogger.protocol.opc.milo.ReadWriteCustomDataTypeNodeExample.main(ReadWriteCustomDataTypeNodeExample.java:52)
```
In configBuilder i have extended Encoding Limits with this follow code: 

```
EncodingLimits limits = new EncodingLimits(EncodingLimits.DEFAULT_MAX_CHUNK_SIZE
				, EncodingLimits.DEFAULT_MAX_CHUNK_COUNT
				, 1535544340
				, EncodingLimits.DEFAULT_MAX_RECURSION_DEPTH);

OpcUaClient opcUaClient = OpcUaClient.create(
	    		  endpointUrl,
	          endpoints ->
	             endpoints.stream()
	                .findFirst(),
	            configBuilder ->
	                configBuilder
	                    .setRequestTimeout(UInteger.valueOf(5000))
	                    .setEncodingLimits(limits)
	                    .build());
```     
Where am i wrong?

Thanks

kevinherron, 2022-12-19:
I don't know, it seems structure is encoded differently than you are expecting. 

Is this structure defined in another server? Do you have to manually define a decoder or does the server define it in a DataTypeDictionary that will allow Milo to dynamically decode it?

kevinherron, 2022-12-19:
This looks like it could be an issue:
```
			Integer DataType = decoder.readByte().intValue();  //OK DECODED --> 2
			System.out.println("DataType=" + DataType + "|");
```
You are decode a single byte, but if the name of the field is any indication to its type you should be decoding an Int32 or something.

You need to figure out a way to see the actual structure definition, including the DataType of each field, not just look at UaExpert's decoded structure value.

EnRiX84, 2022-12-19:
Yep, this structure is defined in another server! The server not define the structure in a DataTypeDictionary unfortunately. 

Your tips are very important, I spoke to the manufacturer of the machine (welding robot) but he has no idea how the opcua server works, the only structure I see is the one with uaExpert unfortunately. 

I tried to figure out the primitive type of data contained within each field but my search was unsuccessful.

[image]

Great! I have decoded the string using 
```
String Name = decoder.readString();
			System.out.println("Name=" + Name + "|");
			
			Integer DataType = decoder.readInt32();
			System.out.println("DataType=" + DataType + "|");
			
			String StringValue = decoder.readString();
			System.out.println("StringValue=" + StringValue + "|");
			
			Integer IntValue = decoder.readByte().intValue();
			System.out.println("IntValue=" + IntValue + "|");
			
			Float FloatValue = decoder.readByte().floatValue();
			System.out.println("Float=" + FloatValue + "|");
```

as you indicated to me.

Thanks a lot

closed 2022-12-19 by kevinherron