eclipse-milo/milo #1093: How to write an ExtensionObject of ExtensionObject[] array

EnRiX84, 2022-12-20:
I'm using milo 0.6.8 and I have decoded the information inside an ExtensionObject of ExtensionObject[] array. 

But now i don't able to write a single location of array. In the follow example i have decoded location 0.

[image]

Following the code of run method of ReadWriteCustomDataTupeNodeExample: 
```

@Override
	public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
		// synchronous connect
		client.connect().get();
		
		// synchronous read request via VariableNode
		UaVariableNode node = client.getAddressSpace().getVariableNode(new NodeId(2, nodeid));
		// Read the current value
		Variant variant = node.readValue().getValue();

		ExtensionObject[] oVal = (ExtensionObject[]) variant.getValue();
		
		for (Object object : oVal) {
			if (object instanceof ExtensionObject) {
				ExtensionObject eo = (ExtensionObject) object;
				NodeId encodingId = eo.getEncodingId();
				registerCustomCodec(client, encodingId);
			}
		}
		
		Object val = oVal[0];
//		Object val = variant.getValue();		
		if (val instanceof ExtensionObject) {
			ExtensionObject eo = (ExtensionObject) val;
			
			CustomStructType decoded = (CustomStructType) eo.decode(client.getStaticSerializationContext());
			System.out.println("Decoded=" + decoded);
			
	        // Write a modified value
			ExtensionObject modifiedXo = ExtensionObject.encode(
	              client.getStaticSerializationContext(),
	              new CustomStructType("Variable", 2, "Magnifico", null, null),
	              eo.getEncodingId(),
	              OpcUaDefaultBinaryEncoding.getInstance());
			
	        node.writeValue(new DataValue(new Variant(modifiedXo)));
	        
	        variant = node.readValue().getValue();
	        eo = (ExtensionObject) variant.getValue();

	        decoded = (CustomStructType) eo.decode(
	            client.getStaticSerializationContext()
	        );

	        System.out.println("Decoded after write=" +  decoded);
		}
		
		future.complete(client);
	}
```

My codec class:

```
public static class Codec implements OpcUaBinaryDataTypeCodec<CustomStructType> {

		@Override
		public Class<CustomStructType> getType() {
			return CustomStructType.class;
		}

		@Override
		public CustomStructType decode(SerializationContext context, OpcUaBinaryStreamDecoder decoder) throws UaSerializationException {
			String Name = decoder.readString();
			Integer DataType = decoder.readInt32();
			String StringValue = decoder.readString();
			Integer IntValue = decoder.readInt32();
			Float FloatValue = decoder.readFloat();
			return new CustomStructType(Name, DataType, StringValue, IntValue, FloatValue);
		}

		@Override
		public void encode(SerializationContext context, OpcUaBinaryStreamEncoder encoder, CustomStructType value) throws UaSerializationException {
			System.out.println("Name=" + value.Name);
			encoder.writeString("Name", value.Name);

			System.out.println("DataType=" + value.DataType);
			encoder.writeInt32("DataType", value.DataType);
			
			System.out.println("StringValue=" + value.StringValue);
			encoder.writeString("StringValue", value.StringValue);
			
			System.out.println("IntValue=" + value.IntValue);
			encoder.writeInt32("IntValue", value.IntValue);
			
			System.out.println("FloatValue=" + value.FloatValue);
			encoder.writeFloat("FloatValue", value.FloatValue);
		}
	}
```

and the result:
[image]

kevinherron, 2022-12-20:
Catch the exception and log the StatusCode associated with it.

I would guess the server is rejecting the write because you are including StatusCode and timestamps with the value, and it does not support writing these:
>```
>node.writeValue(new DataValue(new Variant(modifiedXo)));
>```

The StatusCode associated with the UaException should provide more detail.

kevinherron, 2022-12-20:
It’s not the same object, you are trying to cast `ExtensionObject[]` to `ExtensionObject` somewhere.

EnRiX84, 2022-12-20:
It's true! but to write. I have to write entire ExtensionObject[] with modified location?

kevinherron, 2022-12-20:
That may be required, or if the server supports the “IndexRange” parameter to the Write service you can try that as well. You’ll have to use one of the `OpcUaClient::write` methods to specify that, I think.

Your cast error happens after the write fails maybe:
```
variant = node.readValue().getValue();
	        eo = (ExtensionObject) variant.getValue();
```

The value is still an array in the server.

EnRiX84, 2022-12-20:
I tried to write entire array, i don't receive errors but the write does not work.

```
@Override
	public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
		// synchronous connect
		client.connect().get();
		
		// synchronous read request via VariableNode
		UaVariableNode node = client.getAddressSpace().getVariableNode(new NodeId(2, nodeid));
		// Read the current value
		Variant variant = node.readValue().getValue();
		
		ExtensionObject[] extensionObjectArray = (ExtensionObject[]) variant.getValue();
		
		NodeId encodingId = null;
		for (Object object : extensionObjectArray) {
			if (object instanceof ExtensionObject) {
				ExtensionObject eo = (ExtensionObject) object;
				encodingId = eo.getEncodingId();
				registerCustomCodec(client, encodingId);
			}
		}
		
		ExtensionObject extensionObject = extensionObjectArray[0];
		
		CustomStructType decoded = (CustomStructType) extensionObject.decode(client.getStaticSerializationContext());
		
		System.out.println("Decoded=" + decoded);
		
		decoded.setStringValue("New Text for String");
		
		ExtensionObject modifiedXo = ExtensionObject.encode(
	              client.getStaticSerializationContext(),
	              decoded,
	              extensionObject.getEncodingId(),
	              OpcUaDefaultBinaryEncoding.getInstance());
		
		CustomStructType decoded1 = (CustomStructType) modifiedXo.decode(client.getStaticSerializationContext());
		System.out.println("Decoded1=" + decoded1);
		
		extensionObjectArray[0] = modifiedXo;
		
		CustomStructType decoded2 = (CustomStructType) extensionObjectArray[0].decode(client.getStaticSerializationContext());
		System.out.println("Decoded2=" + decoded2);
		
		client.writeValue(node.getNodeId(), new DataValue(new Variant(extensionObjectArray)));

		future.complete(client);
	}
```

[image]

I think to follow your suggestions but i don't able to write this array... please help me! :)

EnRiX84, 2022-12-20:
I added this code:

```
CompletableFuture<StatusCode> writeValue = client.writeValue(node.getNodeId(), new DataValue(new Variant(extensionObjectArray)));
		
		writeValue.whenCompleteAsync((c, ex) -> {
			if (ex != null) {
				System.out.println("Error running example:"+ ex);
			}
			System.out.println(c);
		});
```

and receive this message: `StatusCode{name=Bad_WriteNotSupported, value=0x80730000, quality=bad}` Also if i try to write single ExtensionObject

kevinherron, 2022-12-20:
Ok, so the server is telling you `Bad_WriteNotSupported`.

Try writing _just_ the value, by creating the DataValue like this:
```
DataValue.valueOnly(new Variant(extensionObjectArray));
```

EnRiX84, 2022-12-20:
WORKS!!!!! THANK YOU!

closed 2022-12-20 by kevinherron