eclipse-milo/milo #1095: Small subscriptions are fine, I subscribe to 30,000 points and somehow don't return subscription data

CH544123735, 2022-12-28:
Dear teachers, I would like to ask you a question.
Small subscriptions are fine, I subscribe to 30,000 points and somehow don't return subscription data.
No error reported

CH544123735, 2022-12-28:
There is no subscription available for this session.

CH544123735, 2022-12-28:
onStatusChanged
2022-12-28 15:19:28.467  WARN 5184 --- [d-thread-pool-4] c.j.o.client.CustomSubscriptionListener  : status=Bad_NoSubscription, description=There is no subscription available for this session.
2022-12-28 15:19:28.471  WARN 5184 --- [d-thread-pool-4] c.j.o.client.CustomSubscriptionListener  : status=Bad_NoSubscription, description=There is no subscription available for this session.
2022-12-28 15:19:28.471  WARN 5184 --- [d-thread-pool-4] c.j.o.client.CustomSubscriptionListener  : UaServiceFaultException: status=Bad_NoSubscription, message=There is no subscription available for this session.
2022-12-28 15:19:28.472  WARN 5184 --- [d-thread-pool-4] c.j.o.client.CustomSubscriptionListener  : onPublishFailure

kevinherron, 2022-12-29:
Maybe the server is taking a long time to create the monitored items and the subscription is timing out while that happens.

If you could get a Wireshark capture that would be useful. You could also try creating the monitored items in smaller batches.

CH544123735, 2022-12-29:
50 points is a group ，There are 2,000 points。 
  Created with "createMonitoredItems"  
  All monitored points are collected once and errors are reported

CH544123735, 2022-12-29:
[image]

closed 2023-02-08 by kevinherron

AK081KA, 2023-06-03:
how to solve this question?