eclipse-milo/milo #1098: Jakarta EE compatible release

oscerd, 2023-01-13:
Hi

Apache Camel has a Milo component and we are working on Camel v4 which is upgrading to JEE 10 (javax -> jakarta). We are using Milo 0.6.8 and with it we are getting:

```
java.lang.NoClassDefFoundError: javax/xml/bind/DatatypeConverter
	at org.eclipse.milo.opcua.stack.core.serialization.OpcUaXmlStreamDecoder.readByteString(OpcUaXmlStreamDecoder.java:332)
	at org.eclipse.milo.opcua.stack.core.serialization.OpcUaXmlStreamDecoder.readBuiltinType(OpcUaXmlStreamDecoder.java:723)
	at org.eclipse.milo.opcua.stack.core.serialization.OpcUaXmlStreamDecoder.readVariantValue(OpcUaXmlStreamDecoder.java:685)
```

Do you plan to do a release that supports Jakarta EE spec ?

kevinherron, 2023-01-13:
This has already been done in the `dev/1.0` branch but unfortunately this release has no target date and a fair amount of work left to do because in addition to API stability it’s also targeting OPC UA 1.05.

oscerd, 2023-01-13:
Thank you, didn't check the dev/1.0 branch. Thanks for the information.

proggler23, 2023-01-17:
hi guys! first of all: nice work, that you are doing here!

sceond of all: could you estimate, when `dev/1.0` will be released (this/next month, in 3 months,...), or does there exist the possibility to cherry-pick the required jakarta-changes to a minor release?

kevinherron, 2023-01-18:
Probably more like next 6 months, it’s hard to say right now.

It can’t be cherry-picked or back-ported because it breaks compatibility in both API/package and required JDK version (I think - Milo currently requires min JDK 8, but moves to min JDK 11 in `dev/1.0`).

ASunc, 2023-02-28:
Are there snapshots of dev/1.0 available anywhere ? Those would be just great for us now, as we are migrating our product to jakartafied modules currently.

closed 2023-04-07 by kevinherron

kevinherron, 2023-06-12:
@frehov https://github.com/eclipse/milo/issues/1130