eclipse-milo/milo #1101: when the opc ua server quality stamp is bad, but the StatusCode in the DataValue received by the opc ua client is good.

armandxu, 2023-01-29:
hi，kevinherron，thank you for your reply , but why when the opc ua server quality stamp is bad, but the StatusCode in the DataValue received by the opc ua client is good.
[image]

[image]

_Originally posted by @armandxu in https://github.com/eclipse/milo/discussions/1100#discussioncomment-4801855_

kimim, 2023-01-29:
As indicated in code:

```java
    private static String quality(StatusCode statusCode) {
        if (statusCode.isGood()) {
            return "good";
        } else if (statusCode.isBad()) {
            return "bad";
        } else if (statusCode.isUncertain()) {
            return "uncertain";
        } else {
            return "unknown";
        }
}
```

quality is just a string name for status code name.

I guess there is another field, you can get quality stamp from. But not from StatusCode.

armandxu, 2023-01-29:
> As indicated in code: ```java private static String quality(StatusCode statusCode) { if (statusCode.isGood()) { return "good"; } else if (statusCode.isBad()) { return "bad"; } else if [...]

I also don't think it's in StatusCode, but I can't find it in other components either.

kevinherron, 2023-01-29:
I don’t know what I’m looking at in the UaExpert? screenshot, but are you sure you’re reading the same NodeId and the same Attribute? It looks like you might be reading DisplayName or Description in the 2nd picture instead of Value.

armandxu, 2023-01-31:
> I don’t know what I’m looking at in the UaExpert? screenshot, but are you sure you’re reading the same NodeId and the same Attribute? It looks like you might be reading DisplayName or Description in the 2nd picture instead of Value.

armandxu, 2023-01-31:
> re you sure you’re reading the same NodeId and the

> I don’t know what I’m looking at in the UaExpert? screenshot, but are you sure you’re reading the same NodeId and the same Attribute? It looks like you might be reading DisplayName or Description in the 2nd picture instead of Value.

hi，kevin，

armandxu, 2023-02-01:
I found the problem, it is a bug in the opc server

closed 2023-02-01 by armandxu