eclipse-milo/milo #1109: The matrix data in the writing operation cannot be set to a one-dimensional array

kdai319, 2023-03-28:
**Describe the bug**
When calling the write method on the client, changing a multidimensional array to a one-dimensional array may cause the current RPC request to fail. In this case the StatusCode.name will return “StatusCode{name=Bad_TypeMismatch, value=0x80740000, quality=bad}”

**Expected behavior**
in the uaExpert client, the same operation for the same node can be successful
Below is an operation screenshot of uaExpert ：
[image]
This is the original state of the node.
[image]
In UaExpert, you can set the dimension of the matrix, but I can't find the setting in milo's writing method. I'm not sure if I missed something, and there doesn't seem to be a similar case in the related tests...
[image]
After setting the dimension as shown in the picture, the written data becomes a one-dimensional array.
[image]
After successful writing, the node state will be updated to the correct form.

**Logs and Packet Captures**
Here is a snippet of my test program:
```
@Test
public void writeMatrixValue() throws Exception {
    OpcUaClient client = getOpcUaClient("opc.tcp://WIN-J3CRL509SHS:48010");
    NodeId nodeId = new NodeId(2, "Demo.Static.Matrix.Byte");
    DataValue dataValue = new DataValue(new Variant(
            new UByte[][]{
                    new UByte[]{UByte.valueOf(11)}
            }
    ), StatusCode.GOOD, null);
    StatusCode statusCode = client.writeValue(nodeId, dataValue).get();
    log.debug("{}", statusCode);
}
```
If you write a two-dimensional array just like in the program, it can be successful.
[image]

But if you change the value to be written to a one-dimensional array, you will get the error mentioned before. like this:
```
DataValue dataValue = new DataValue(new Variant(
        //new UByte[][]{
                new UByte[]{UByte.valueOf(11)}
        //}
), StatusCode.GOOD, null);
```

**Additional context**
The tested server program is UaCPPServer from Unified Automation.
The milo library used is the latest version 0.6.8.

kevinherron, 2023-03-28:
Do you have Wireshark captures that can be compared? That seems like the easiest approach to figure out what is different.

kdai319, 2023-03-28:
Okay, buddy.
These are wireshark packages for your reference.
[pcapng.zip](https://github.com/eclipse/milo/files/11091278/pcapng.zip)

kevinherron, 2023-03-28:
I don't know if this can be matched right now. Explicit Matrix support has been added in the `dev/1.0` branch but there isn't an exact equivalent in the current releases. 

Right now it looks like you would have to pass in what you already tried (2 or more dimensional array) in order for the array dimensions to be included and indicated in the encoding mask [0].

Milo:
[image]

UaExpert:
[image]

[0] https://reference.opcfoundation.org/Core/Part6/v105/docs/5.2.2.16

kevinherron, 2023-03-29:
I am also not convinced the server should even accept the write from UaExpert because now the Array Dimensions do not match the value.

closed 2023-03-29 by kevinherron