eclipse-milo/milo #1112: StatusCode{name=Bad_Timeout, value=0x800A0000, quality=bad} on UaSubscription.NotificationListener.super.onStatusChangedNotification

demourarosa38, 2023-04-04:
**Describe the bug**
We have a node OPC with more than 3000 tags.
And frequently after default timeout session (120 seconds), the listner of subscription said:
StatusCode{name=Bad_Timeout, value=0x800A0000, quality=bad}
The tags in this node change more than 90000 per day.

**Expected behavior**
We expect no Bad_Timeout or less than 100 per day.

**Additional context**
We use the version 0.6.6.
We are connected to a VNode server with other nodes, like 30.
And with 2000 tags average per node.

Thanks

kevinherron, 2023-04-04:
Bad_Timeout is the server telling you the subscription has timed out. It's not coming from Milo or the client side.

If your client is creating a lot of subscription it's possible you need to be sending more PublishRequests than is configured by default - you can control this via the [MaxPendingPublishRequests setting](https://github.com/eclipse/milo/blob/3235f91e034d830a409039543113b5d00f626bf5/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/api/config/OpcUaClientConfig.java#L60-L63).

If that's not the issue then you probably need to be troubleshooting this on the server side. It's always going to be helpful if you can record a Wireshark capture leading up to and including when the problem starts.

closed 2023-04-08 by kevinherron

demourarosa38, 2023-04-13:
Hello,

Thanks for the response.
I'd migrated from UaSubscription to ManagedSubscription and i've got less Bad_Timeout with this class.
I suppose that my operations on data changed are big with further request on database.
With ManagedSubscription, my operations execute with batch of datachanges, and it's better.

I think to resolve completly, i have to perform my operations on database.

See you.